KumarRobotics/ublox #8: It is possible to add the online assisted gps service

olmerg, 2015-08-19:
the u-blox receivers are fitted with what the company calls "AssistNow" capability and it does exactly the same thing your iPhone does - feeds the GPS with pre-downloaded almanacs

the form to do it it is documented here:
https://gist.github.com/veproza/55ec6eaa612781ac29e7

versatran01, 2015-08-19:
@olmerg 
This looks interesting. I also have trouble finding 3D fix sometimes.
I will take a look at it.

olmerg, 2015-08-19:
ok.
My university block the service but I put here because  that will be increase the accuracy.

vmlane, 2017-07-25:
you're welcome to add this functionality to the repo and submit a pull request. You would need to add MGA messages and a handler for them,

andresR8, 2018-04-27:
I've opened a PR with my implementation https://github.com/KumarRobotics/ublox/pull/31