KumarRobotics/ublox #12: Cannot enable/disable sbas on ublox M8P

zacwitte, 2016-12-16:
Starting the node always fails with the message `Error configuring device: Failed to disable SBAS.` (or enabling it - neither works).

We have a Ublox M8P connected via USB. I've tried 9600 and 4800 baud rate.

kartikmohta, 2016-12-16:
You can try commenting out the code applying the SBAS setting (https://github.com/KumarRobotics/ublox/blob/fa26d5548b460881c47da6f50b1e6dcadd2a4b1f/ublox_gps/src/node.cpp#L420-L424)

vmlane, 2017-07-18:
I recommend this issue be closed since it's been fixed with the latest merge. The issue was this: M8P does not have SBAS capabilities. When the SBAS message was sent, the device would return a NACK. The program throws a runtime exception if it receives a NACK during configuration, thus the error.

The latest commit verifies that the device is SBAS capable by parsing the MonVER message and only sends CfgSBAS messages if the device is SBAS capable.

closed 2017-07-21 by kartikmohta