KumarRobotics/ublox #15: Error Message "Failed to set measurement rate to xx ms"

philippebrand, 2017-05-30:
Hello

I have a low-cost RTK-GPS with an U-BLOX M8T Chip onboard.

In the output-Stream should come only the following messages:

- NAV-POSLLH
- NAV-STATUS
- NAV-DOP
- NAV-VELNED
- NAV-SOL 

The outputstream comes in the following format:

Rate:      5 Hz
Format:**38400:8:n:1:off**
-> baudrate 38400, 8 bits, no parity check, 1 stop bit, no flow control

I adapted the launch-file with these parameters, but I always get the following Error-Message:

**Failed to set measurement rate to 200ms**

So I tried wiht other rates (1-16Hz), but that never worked.

If you can give me a tip where the problem is, that would be very helpful.

Thank you for the answer.

kartikmohta, 2017-05-30:
Just for a basic check, have you verified that 38400 is the correct baud rate for your unit? The default ublox baud rate is 9600.

philippebrand, 2017-05-31:
Hello, thank your for the fast answer!

Yes, I tried also to set the baudrate to 9600, but also with this configuration I get the Error Message.
The Receiver I use is the following:

https://drotek.com/en/documentation/smartnav-rtk-documentation/

FYI on this page under -> _Advanced Users_ -> _Manual Configuration_ , is the output of the ubx-stream described.

kartikmohta, 2017-05-31:
Looks like there is no direct connection to the ublox GPS module on that board. The port with baud rate 38400 is a UART port of the Intel Edison so I don't think changing the ublox settings would work with that port. To change the ublox settings, you would need to connect directly to the ublox module as shown here: https://drotek.com/en/documentation/smartnav-rtk-documentation/#user-content-configuring-u-blox-chips

philippebrand, 2017-06-01:
Yes, that could be the problem. Unfortunately I have at the moment no adapter for a direct connection to the chip. So I think it's not that easy to solve this problem very fast. Actually the Rover should be able to send NMEA-Messages. Then I would use the _nmea_navsat_driver_. But with this configuration I have also some problems (seems like there comes no data with this configuration).

Thank you for the help and I will inform you when I have a solution.

vmlane, 2017-07-25:
This issue has been fixed version 1.1.0, the baudrate was not configured properly.

closed 2017-07-25 by vmlane