KumarRobotics/ublox #20: U-Blox ASIO input buffer read error: End of file, 0

innavalieva, 2017-08-09:
Hi!
I am getting the following error all the time:
U-Blox ASIO input buffer read error: End of file, 0
Could you recommend how to resolve this issue?

Best regards,
Inna

kartikmohta, 2017-08-11:
Try adding the following block on line 118 (i.e. after the serial port is opened) in ublox_gps/src/gps.cpp:
```cpp
{
  // Wait for at least one char on serial port before returning from read.
  // Sadly Boost doesn't provide any way to set this.
  int fd = serial_port->native_handle();
  struct termios tio;
  tcgetattr(fd, &tio);
  tio.c_cc[VTIME] = 0;
  tio.c_cc[VMIN] = 1;
  tcflush(fd, TCIFLUSH);
  tcsetattr(fd, TCSANOW, &tio);
}
```

innavalieva, 2017-08-11:
Thank you for your suggestion Kartik! 
I have added the block on line 118 in ublox_gps/src/gps.cpp as you suggestaed:

` // open serial port
  try {
    serial->open(port);
  } catch (std::runtime_error& e) {
    throw std::runtime_error("U-Blox: Could not open serial port :" 
                             + port + " " + e.what());
  }

  ROS_INFO("U-Blox: Opened serial port %s", port.c_str());

 //ADDED ON LINE 118 TO FIX ASIO input problem
  {
    // Wait for at least one char on serial port before returning from read.
    // Sadly Boost doesn't provide any way to set this.
    int fd = serial_port->native_handle();
    struct termios tio;
    tcgetattr(fd, &tio);
    tio.c_cc[VTIME] = 0;
    tio.c_cc[VMIN] = 1;
    tcflush(fd, TCIFLUSH);
    tcsetattr(fd, TCSANOW, &tio);
  }
// End of added  ON LINE 118 TO FIX ASIO input problem

  // Set the I/O worker
  if (worker_) return;
  setWorker(boost::shared_ptr<Worker>(
   new AsyncWorker<boost::asio::serial_port>(serial, io_service)));

` // open serial port
  try {
    serial->open(port);
  } catch (std::runtime_error& e) {
    throw std::runtime_error("U-Blox: Could not open serial port :" 
                             + port + " " + e.what());
  }

  ROS_INFO("U-Blox: Opened serial port %s", port.c_str());
 //ADDED ON LINE 118 TO FIX ASIO input problem
  {
    // Wait for at least one char on serial port before returning from read.
    // Sadly Boost doesn't provide any way to set this.
    int fd = serial_port->native_handle();
    struct termios tio;
    tcgetattr(fd, &tio);
    tio.c_cc[VTIME] = 0;
    tio.c_cc[VMIN] = 1;
    tcflush(fd, TCIFLUSH);
    tcsetattr(fd, TCSANOW, &tio);
  }
// End of added  ON LINE 118 TO FIX ASIO input problem

  // Set the I/O worker
  if (worker_) return;
  setWorker(boost::shared_ptr<Worker>(
   new AsyncWorker<boost::asio::serial_port>(serial, io_service)));
`
However the error still exists. 

Best regards,
Inna

kartikmohta, 2017-08-11:
Did you recompile after adding that code? In its current form, it would give you compile errors. You have to change `serial_port->native_handle();` to `serial->native_handle();` and the recompile.

pkess, 2017-09-26:
Hi there,
i got the same issue on a raspberry pi with a ros image installed (http://www.german-robot.com/2016/05/26/raspberry-pi-sd-card-image/)

Did anyone solve this?

Additionally i found out that the usb-device disconnects when the code tried to set dtr/rts signals:
dmesg:
`[ 1361.535370] usb 1-1.2: USB disconnect, device number 28
[ 1361.535543] cdc_acm 1-1.2:1.0: failed to set dtr/rts
`

weisongwen, 2017-11-09:
Hi, @pkess @kartikmohta  
Have you solved this problem, I also get this issue when I use this driver to get data from EVK-M8T. But I can get the data when i use the u-center in windows 10.

Thanks

pkess, 2017-11-10:
Hi there,
Unfortunately I could not solve this issue, so I decided to start again with a clean raspbian Jessy image and compiled Ros from source. Then tge error was gone.
Took about one day to get all dependencies compiled

weisongwen, 2017-11-13:
Hi, @pkess ， 
Thanks for your reply. you mean that this is a problem casued by some dependencies needed?

weisongwen, 2017-11-13:
Hi, @innavalieva  
Have you solved this problem, I also get this issue when I use this driver to get data from EVK-M8T. But I can get the data when i use the u-center in windows 10.

billyjov, 2018-02-08:
does someone got this driver working on any device?

kartikmohta, 2018-02-08:
Can you try with the latest commit? That should help with the end of file issue.
There are still some issues with devices with timing features such as the M8T which need to be looked at.

billyjov, 2018-02-08:
yeah i will try it in a few hours and give a feedback.  

BTW  i use the `GPS U-blox NEO-6M ``.

weisongwen, 2018-02-09:
Hi @kartikmohta  
I still cannot get data from M8T with driver. just as you said, this driver is not available for M8T by now,

Best 
Welson

pec-orange, 2018-05-18:
I had this one with NEO-M8P:
U-Blox ASIO input buffer read error: End of file, 0

Solved it by changing lines 419-421 in node.cpp to
if (false) {
       gps.configUsb(usb_tx_, usb_in_, usb_out_);
}

It seems like my node was trying to connect to the device by uart and usb simultaneously. And it was failing since I didn't specify any usb connection parameters in the config file.
Anyway, this is how I fixed it.

P.S.
If it doesn't help - you may also try to remove the corresponding lines in gps.configUsb() function itself. It is located in gps.cpp.

siddharthcb, 2021-09-13:
> Thank you for your suggestion Kartik!
> I have added the block on line 118 in ublox_gps/src/gps.cpp as you suggestaed:
> 
> ` // open serial port
> try {
> serial->open(port);
> } catch (std::runtime_error& e) {
> throw std::runtime_error("U-Blox: Could not open serial port :"
> + port + " " + e.what());
> }
> 
> ROS_INFO("U-Blox: Opened serial port %s", port.c_str());
> 
> //ADDED ON LINE 118 TO FIX ASIO input problem
> {
> // Wait for at least one char on serial port before returning from read.
> // Sadly Boost doesn't provide any way to set this.
> int fd = serial_port->native_handle();
> struct termios tio;
> tcgetattr(fd, &tio);
> tio.c_cc[VTIME] = 0;
> tio.c_cc[VMIN] = 1;
> tcflush(fd, TCIFLUSH);
> tcsetattr(fd, TCSANOW, &tio);
> }
> // End of added ON LINE 118 TO FIX ASIO input problem
> 
> // Set the I/O worker
> if (worker_) return;
> setWorker(boost::shared_ptr(
> new AsyncWorkerboost::asio::serial_port(serial, io_service)));
> 
> ` // open serial port
> try {
> serial->open(port);
> } catch (std::runtime_error& e) {
> throw std::runtime_error("U-Blox: Could not open serial port :"
> + port + " " + e.what());
> }
> 
> ROS_INFO("U-Blox: Opened serial port %s", port.c_str());
> //ADDED ON LINE 118 TO FIX ASIO input problem
> {
> // Wait for at least one char on serial port before returning from read.
> // Sadly Boost doesn't provide any way to set this.
> int fd = serial_port->native_handle();
> struct termios tio;
> tcgetattr(fd, &tio);
> tio.c_cc[VTIME] = 0;
> tio.c_cc[VMIN] = 1;
> tcflush(fd, TCIFLUSH);
> tcsetattr(fd, TCSANOW, &tio);
> }
> // End of added ON LINE 118 TO FIX ASIO input problem
> 
> // Set the I/O worker
> if (worker_) return;
> setWorker(boost::shared_ptr(
> new AsyncWorkerboost::asio::serial_port(serial, io_service)));
> `
> However the error still exists.
> 
> Best regards,
> [image]
> 
> Inna

had the same error when connected through serial on raspi 4..

closed 2022-08-02 by fcladera