KumarRobotics/ublox #22: U-Blox ASIO input buffer read error: Operation canceled, collecting data from ublox EVK-M8T

weisongwen, 2017-11-09:
Hi,
when I use this ublox driver to collect data from ublox EVK-M8T using the mini-USB port, I get the error as following 
[image]
my launch file is 
`<?xml version="1.0" encoding="UTF-8"?>

<launch>
  <arg name="output" default="screen" />
  <arg name="respawn" default="true" />
  <arg name="respawn_delay" default="30" />
  <arg name="clear_params" default="true" />

  <node pkg="ublox_gps" type="ublox_gps" name="ublox_gp"
        output="$(arg output)" 
        clear_params="$(arg clear_params)"
        respawn="$(arg respawn)" 
        respawn_delay="$(arg respawn_delay)">
    <rosparam command="load" 
              file="$(find ublox_gps)/config/nmea.yaml" />
  </node>
</launch>
`
**Operation canceled, collecting data from ublox EVK-M8T** this seems to be the main error. I have tried the method mentioned in [here](https://github.com/KumarRobotics/ublox/issues/20)https://github.com/KumarRobotics/ublox/issues/20, but it does not work.

do you have any advice on this error?

weisongwen, 2017-11-13:
Hi, @kartikmohta 
sorry to bother you here, do you have any advice on this issue?

kartikmohta, 2017-12-08:
Hi,
Can you try the latest version from the repo?
If that does not fix the problem, can you run the ublox_gps node under gdb following the instructions at http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB and paste the backtrace from gdb when it crashes? (You can set respawn to false for this).

Kortium, 2017-12-10:
Run into same issue here's what i got from the gdb
[image]

May be it's because i run it on ARM linux

kartikmohta, 2017-12-10:
@Kortium Can you provide the backtrace from gdb? That may help in debugging where the problem is occurring.

Kortium, 2017-12-10:
Here it is

[image]

weisongwen, 2017-12-11:
Hi @kartikmohta @Kortium , I am so glad to get both of your suggestions. I get such issue again.
I debugged this error, and finally found that this package did not support the Ublox EVK M8T product. It seems that the driver for M8T have not been finished yet.

Best 
Welson

kartikmohta, 2017-12-12:
@Kortium @weisongwen Can you try again with the latest version from the master branch? Commit 4570130ad5adae4add35feda10a806b6d6d7ad9d fixed a bug which could have led to segfaults.

weisongwen, 2017-12-12:
Hi @kartikmohta  Thanks for your effort. I will try this first.

Kortium, 2017-12-12:
My results:

[image]

weisongwen, 2017-12-12:
Hi， @Kortium 
Thanks for your share of your result. 
I want to confirm one thing: the device you used is ublox EVK M8T?

Kortium, 2017-12-12:
Hi, @weisongwen 
That's a good question. No my device is C94-M8P-3 (devkit). But issue looks the same...

weisongwen, 2017-12-12:
Hi, @kartikmohta   Thanks for your suggestion.
I tested the new package again and I find that issue again.
also I got the warning from the node.h file:
```
void getRosParams() {
    ROS_WARN("Functionality specific to u-blox TIM devices is only %s",
        "partially implemented. See TimProduct class in node.h & node.cpp.");
  }
```
It seems that this package do not support the ublox EVK M8T device?

Best 
Welson

kartikmohta, 2017-12-12:
@Kortium It seems like the resetting is not working properly for your device, can you try enabling the gnss/qzss option in the param file and try again? From your output, looks like the QZSS Configuration is the only one causing a GNSS re-configuration.

@weisongwen I think a similar problem for you, can you enable the gnss/glonass option in the param file and retry?

Kortium, 2017-12-12:
@kartikmohta @weisongwen 
Thanks a lot for opening this question and helping me, i have misformed config file ~.~
Now it works!

weisongwen, 2017-12-13:
Hi，@kartikmohta 
Thanks for your kind reply. Autally, I also have a M8N in hand. this driver works well  with M8N. But not compatable with M8T .

I will test the M8T as your advice. Thanks!

Best 
Welson

weisongwen, 2018-01-26:
Hi @Kortium 
Can you share your config file to me ? Thanks a lot.

weisongwen, 2018-01-26:
Dear @kartikmohta 
Thanks for your effort on maintaining this package, It is mentioned in the Readme part:
``
For FTS & TIM devices:

    currently unimplemented. See FtsProduct and TimProduct classes in ublox_gps package node.h & node.cpp files.

``
does TIM devices means the Ublox M8T ?

Best 
Welson

kartikmohta, 2018-02-08:
@weisongwen TIM means devices with timing feature (https://www.u-blox.com/en/timing) so M8T devices are not supported right now.

benjaminsamsonnettr, 2018-04-06:
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  write: Input/output error
Aborted (core dumped)

could you please help me

pec-orange, 2018-05-18:
I also had this error:
U-Blox ASIO input buffer read error: Operation canceled, 0

Solved it by changing tmode3 to 0 in the config file...

MrBanannaMan, 2018-06-12:
For those using the M8T modules, please use this gentleman's modified repository:
https://github.com/flynneva/ublox.git2

After much frustration with my M8T module, I found this repo to work wonderfully.

weisongwen, 2018-06-13:
hi @MrBanannaMan,
thanks for your information, but this https://github.com/flynneva/ublox.git2 can not be opened. seems that this package is not existed.

MrBanannaMan, 2018-06-13:
@weisongwen 
Hi Welson,
My apologies, a "2" snuck it's way into that link at the very end. Try this:

https://github.com/flynneva/ublox

weisongwen, 2018-06-14:
hi @MrBanannaMan  
thanks for your kind information, 
best 
welson

WillenPeng, 2018-06-14:
@weisongwen Hi, did you find good config file for m8t?

weisongwen, 2018-06-25:
@MrBanannaMan  Hi, have you find that where we can obtain the pseudorange measurements from this package? moreover, this package seems cannot work with the Beidou satellites.

best
welson

flynneva, 2018-06-30:
Hi Welson,

Take a look at the config file you are using to launch the m8t. It should be in the  'ublox_gps/config/' directory....probably set to the 'c94_m8t_base.yaml'.

According to the [wiki for this ublox package](http://wiki.ros.org/ublox)......Beidou satellites are default set to off.

Add the following line to your yaml file and it should work:
```
gnss:
  beidou: true
```