KumarRobotics/ublox #24: Possible to see GPS status?

matteopantano, 2017-12-31:
I have a c94 development board and I am running ROS on the rover module using the config file provided by this repo.
`gps/fix` tell me just the latitude and longitude 
I am guessing that `gps/navstatus` tell me something that could be useful for me but I cannot understand the meaning of the message:
```
iTOW: 60582000
gpsFix: 3
flags: 223
fixStat: 1
flags2: 8
ttff: 32534
msss: 512530
```
Can someone help me telling me what should I look for having a confirmation of fixed rtk gps signal? Which topic should I look to see if the GPS has a fixed signal?

Thanks

kartikmohta, 2017-12-31:
The ROS messages are just a copy of the UBX messages sent by the GPS module. You can find more information about the contents of the messages in the UBX protocol specification section of their [manual](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29.pdf).

matteopantano, 2018-01-04:
Thanks that was really helpful!

closed 2018-01-04 by matteopantano