KumarRobotics/ublox #25: any one successfully employ this driver in ublox M8T device?

weisongwen, 2018-01-24:
Dear All 
Thanks for reading this pacakge.  do any one successfully employ this driver in ublox M8T device?

Thanks 
Welson

kasraghu, 2018-03-06:
Used it by changing [this line](https://github.com/KumarRobotics/ublox/blob/master/ublox_gps/include/ublox_gps/node.h#L1291) to `bool configureUblox() { return true; }`

weisongwen, 2018-03-07:
Hi, @kasraghu  
Thanks for your kind information. Have you test this with M8T?

Thanks 
Welson

kasraghu, 2018-03-07:
I have tested only SFRBX and RAWX messages from an M8T.

benjaminsamsonnettr, 2018-04-04:
i have done the above config nut facing the errors like this

[ERROR] [1522832692.742706954]: U-Blox ASIO input buffer read error: End of file, 0
[ERROR] [1522832692.742842816]: U-Blox ASIO input buffer read error: End of file, 0
[ERROR] [1522832692.742974890]: U-Blox ASIO input buffer read error: End of file, 0
[ERROR] [1522832692.743107968]: U-Blox ASIO input buffer read error: End of file, 0

any suggestions ?

flynneva, 2018-04-17:
I've submitted a pull request with the additions to make this library work with the M8T device. Check out my repo in the mean time until @Kumar approves the request.

FPSychotic, 2023-02-02:
> I've submitted a pull request with the additions to make this library work with the M8T device. Check out my repo in the mean time until @Kumar approves the request.

Hi , do you know if it was included, I think so, and if it accepts raw data and load the gps configuration without overwrite the setup on flash? I couldn't see any reference to that