KumarRobotics/ublox #26: Running the driver on Raspberry PI

Hasan-ta, 2018-02-14:
I have a raspberry PI loaded with Ubuntu and ROS Kinetic. The driver compiles and runs on the raspberry PI but doesn't publish the requested topics, although the same driver runs fine on x86 based system also with ROS Kinetic. Anyone had a similar issue?

billyjov, 2018-03-29:
@Hasan-ta Yeah I ran into the same issue. But dont know why i got the `status = -1` and i'm be able to read data.

billyjov, 2018-03-29:
[image]
[image]

sidtanwar, 2018-04-11:
I have the same issue as @Hasan-ta . I am using a Ublox LEA-6T receiver with Raspberry Pi and ROS Indigo.  Has someone come up with a solution yet?

mistoll, 2018-08-03:
Should be fixed by e255881

fcladera, 2022-08-03:
Closed as it is a stale issue. Feel free to reopen it if you still are facing this problem.

closed 2022-08-03 by fcladera