KumarRobotics/ublox #29: launch application

benjaminsamsonnettr, 2018-04-04:
suggest me 
1, how to ran the application with parameters ?
2, what is the value to be passed for  node_name 
as per understanding roslaunch launch/ublox_device.launch  node_name = ? 

pls suggest

AchmadFathoni, 2019-06-18:
Hi Benjamin, are you still in trouble launching ros node with parameter?

horitahiroshi, 2020-12-25:
Hi!
I have the same question.
How do I run the application?

I am using Ubuntu 18.04 and ROS melodic.
I have installed this package through catkin_ws, and now I am trying to launch the application.

versatran01, 2020-12-26:
Use this launch file
https://github.com/KumarRobotics/ublox/blob/master/ublox_gps/launch/ublox_device.launch

horitahiroshi, 2020-12-26:
Thanks @versatran01.
I since am usin nmea, I called the application with `param_file_name:=nmea`, and changed the default path of `param_file_dir` according to where the /ublox_gps dir is in my computer.

But something is going wrong.
`started roslaunch server http://horitahiroshi-inspiron:46177/

SUMMARY
========

CLEAR PARAMETERS
 * /ublox/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /ublox/nmea/bds_talker_id: [0, 0]
 * /ublox/nmea/compat: True
 * /ublox/nmea/consider: True
 * /ublox/nmea/filter/date: True
 * /ublox/nmea/filter/gps_only: True
 * /ublox/nmea/filter/msk_pos: True
 * /ublox/nmea/filter/pos: True
 * /ublox/nmea/filter/time: True
 * /ublox/nmea/filter/track: True
 * /ublox/nmea/gnssToFilter/beidou: True
 * /ublox/nmea/gnssToFilter/glonass: False
 * /ublox/nmea/gnssToFilter/gps: False
 * /ublox/nmea/gnssToFilter/qzss: True
 * /ublox/nmea/gnssToFilter/sbas: True
 * /ublox/nmea/gsv_talker_id: 1
 * /ublox/nmea/high_prec: False
 * /ublox/nmea/limit82: True
 * /ublox/nmea/main_talker_id: 1
 * /ublox/nmea/num_sv: 8
 * /ublox/nmea/set: True
 * /ublox/nmea/sv_numbering: 1
 * /ublox/nmea/version: 65

NODES
  /
    ublox (ublox_gps/ublox_gps)

ROS_MASTER_URI=http://localhost:11311

ERROR: cannot launch node of type [ublox_gps/ublox_gps]: ublox_gps
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete`

Although there is data coming from the Ublox module on /dev/ttyACM0.

fcladera, 2022-08-03:
Did you configure the yaml that you are using with `param_file_name`? Closing this issue by now, feel free to reopen it if you still have problems.

closed 2022-08-03 by fcladera