KumarRobotics/ublox #30: RXM message generation for C94-M8P device

alblinks, 2018-04-20:
With the C94-M8P evaluation board it is currenctly not possible to generate RXM messages, even if the receiver is correctly configurated. This is due to a twist in u-blox product description. In the u-blox 8 Protocol Specification [1, p. 337] it says RXM-RAWX messages are only supported on Time Sync products, however in the datasheet for the evaluation board [2, p. 11] it is stated that "The NEO-M8P modules provide raw measurement data [...] in the UBX-RXM-RAWX message". 

In the code for ublox_gps node, rawx is currently only implemented for timing products [here](https://github.com/KumarRobotics/ublox/blob/master/ublox_gps/src/node.cpp#L1580). However, all devices using an u-blox M8P chipset generate raw data using RXM-RAWX instead of RXM-RAW. 

I guess there should be some distinction in [RawDataProduct](https://github.com/KumarRobotics/ublox/blob/694268c3b42434eb9a9102eaaa17f57bc444a310/ublox_gps/src/node.cpp#L1205).

As I have never worked with cpp, I can't suggest a fix. I would be happy about any fix.

Thanks!

[1] https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29_Public.pdf

[2] https://www.u-blox.com/sites/default/files/NEO-M8P_DataSheet_%28UBX-15016656%29.pdf

alblinks, 2018-04-20:
I found a working solution: I just copied the content of [TimProduct::subscribe](https://github.com/KumarRobotics/ublox/blob/694268c3b42434eb9a9102eaaa17f57bc444a310/ublox_gps/src/node.cpp#L1578) to [HpgRefProduct::subscribe](https://github.com/KumarRobotics/ublox/blob/694268c3b42434eb9a9102eaaa17f57bc444a310/ublox_gps/src/node.cpp#L1409) and [HpgRovProduct::subscribe](https://github.com/KumarRobotics/ublox/blob/694268c3b42434eb9a9102eaaa17f57bc444a310/ublox_gps/src/node.cpp#L1513). 

If I'm correct, all u-blox products supporting RTCM output (and thus raw data output) use RAWX instead of RAW and SFRBX insted of SFRB.

idelpino, 2019-03-05:
Thanks! It worked!

aaguiar96, 2022-10-25:
Also worked for me for f9p modules :)