KumarRobotics/ublox #33: The node hangs on the survey-in completion

dvornikov-aa, 2018-05-30:
I'm trying to get RTCM stream from Drotek Tiny RTK GPS (ublox NEO-M8P-2-11). I faced a strange node behavior: the node hangs on the survey-in completion.

Here is a clipping from the log:

> [DEBUG] [1527670615.269908056]: U-Blox sent 18 bytes: 
> b5 62 0b 30 01 00 16 52 10 b5 62 0b 31 01 00 16 53 14 
> [DEBUG] [1527670616.269507921]: U-Blox sent 18 bytes: 
> b5 62 0b 30 01 00 17 53 11 b5 62 0b 31 01 00 17 54 15 
> [ INFO] [1527670617.083385910]: Setting mode (internal state) to Time Mode
> [DEBUG] [1527670617.083520593]: Configuring measurement rate to 1000 and nav rate to 1
> [DEBUG] [1527670617.083661341]: Waiting for ACK 0x06 / 0x08
> [ERROR] [1527671071.268433237]: Ublox AsyncWorker::send: Output buffer too full to send message
> [ERROR] [1527671071.268639170]: Ublox AsyncWorker::send: Output buffer too full to send message

I clarified the stop point:

> Gps::waitForAcknowledge(const boost::posix_time::time_duration& timeout, uint8_t class_id, uint8_t msg_id) -> worker_->wait(timeout) -> ScopedLock lock(read_mutex_)

Here is the configuration file (survey-in settings are set to allow survey-in completion in the office for the debug purposes):
[c94_m8p_base.yaml.gz](https://github.com/KumarRobotics/ublox/files/2053428/c94_m8p_base.yaml.gz)

Could you give a hint on this issue?

Thanks!

Timple, 2018-09-24:
Did you try setting a baudrate of 19200 in the config file?

trixr4kdz, 2019-03-07:
@dvornikov-aa, did you ever find a fix for this? I’m currently having the same problem.