KumarRobotics/ublox #43: feature request: transmit RTCM messages over ROS topics

Timple, 2018-09-28:
Since our robots have already ROS communication setup between the robots and a docking station, it would be convenient if the RTK-GPS messages can be transferred via this communication channel.
Instead of setting up a seperate serial line.

Would this be a useful feature or am I missing something here? 

With respect to latency I think we should be able to get below 500ms which is enough for ublox.

Timple, 2018-10-02:
For completeness: a similar request has been made on ROS answers:
https://answers.ros.org/question/283053/ublox-package-configuration/

Posting it here for future reference.

nicolaje, 2020-02-03:
Hello,

I would definitely be interested in the ability to transmit RTCM messages over a ROS topic to the uBlox driver, which would be in charge of sending them to the board.

Is this something planned ?

shac12, 2020-10-23:
Hi, 

I would also be very interested in this. There are many publicly available NTRIP servers it is just a situation of inputting these correction messages to the GPS module to get cm accuracy outdoors.

Kind regards,
Sharif

Jconn, 2020-12-22:
This was implemented in another repo here: https://github.com/ros-agriculture/ublox_f9p
the README there is instructive on how to get things working.

GaneshGovind1, 2022-06-24:
> Since our robots have already ROS communication setup between the robots and a docking station, it would be convenient if the RTK-GPS messages can be transferred via this communication channel. [...]

Hi, can you help me to connect the ublox with ROS ?