KumarRobotics/ublox #45: Issue with multiple GPS nodes

ArjunBalakrishnan, 2018-12-04:
Hello,

I have two ublox GPS modules and I want to connect and record data from both of them simultaneously. 
I followed[ this](https://github.com/KumarRobotics/ublox/issues/7) suggestion and created a launch file as given below. It creates two different nodes _gps1_ and _gps2_. But topics are only published in one of them. What could be the error? I highly appreciate any suggestions.  

 ```
<launch>
 <arg name="output" default="screen" />
  <arg name="respawn" default="true" />
  <arg name="respawn_delay" default="30" />
  <arg name="clear_params" default="true" />

  <node pkg="ublox_gps" type="ublox_gps" name="gps1"
        output="$(arg output)" 
        clear_params="$(arg clear_params)"
        respawn="$(arg respawn)" 
        respawn_delay="$(arg respawn_delay)">
    <rosparam command="load" 
              file="$(find ublox_gps)/config/gps1.yaml" />
  </node>

  <node pkg="ublox_gps" type="ublox_gps" name="gps2"
        output="$(arg output)" 
        clear_params="$(arg clear_params)"
        respawn="$(arg respawn)" 
        respawn_delay="$(arg respawn_delay)">
    <rosparam command="load" 
              file="$(find ublox_gps)/config/gps2.yaml" />
  </node>
</launch>

```
Thank you

rikba, 2022-05-20:
I'm trying the same and it appears that connecting two receivers causes problems with how the serial polling is done. The messages only appear sparsely on both devices. I could not figure out, yet, what is wrong. I'm connecting via USB.

rikba, 2022-05-20:
Ok, my problem was that I did not set the parameter `device` to the proper `/dev/ttyACM0` and `/dev/ttyACM1`.

closed 2022-08-03 by fcladera