KumarRobotics/ublox #48: Making it work for Protocol Versions 27 eg. ZED-F9P

virraaj, 2019-03-05:
I am trying to make this repository work for ZED-F9P hardware. I am having a C099-F9P board. 
What must be the steps that I must follow in order to make it work. Initially, for first step I am only trying to get the NMEA data. The following is the config file:
debug: 1                    # Range 0-4 (0 means no debug statements will print)
device: /dev/ttyACM0
frame_id: gps
dynamic_model: portable
fix_mode: 2D              # Switches between 2D/3D automatically
dr_limit: 0                 # Dead reckoning limit
enable_ppp: false           # Advanced setting not supported by all devices

rate: 2                     # Measurement rate in Hz
nav_rate: 4                 # in number of measurement cycles

uart1:
  baudrate: 38400           # baudrate is device specific, check the device manual
  in: 1                     # UBX
  out: 2                    # NMEA
gnss:
  gps: true                 # (not required since it defaults to true)
  glonass: true
  beidou: true
  qzss: true
  sbas: false
  galileo: true

nmea:
  set: true
  version: 65
  num_sv: 8
  sv_numbering: 1
  compat: true
  consider: true
  limit82: true
  high_prec: false
  filter:
    pos: false
    msk_pos: false
    time: false
    date: false
    gps_only: false
    track: false
  gnssToFilter:
    gps: false
    sbas: true
    qzss: false
    glonass: false
    beidou: false
  main_talker_id: 1
  gsv_talker_id: [0,0]
  bds_talker_id: [0,0]

inf:
  all: true                   # Whether to display INF messages
publish:
  all: true
  aid:
    hui: false
  tim:
    tm2: true
  nav:
    clock: false
    status: false
subscribe:
  all: true                 # Subscribe to all messages
  aid:
    all: false                # ... except AID messages`

The attached image shows the information I get from launching it. 
[image]

Can anyone guide me, from where should I start editing in order to make it work

MarcoMers, 2019-03-21:
Hi,i'm  a newbie and i have the same problem... have you made any progress or find a solution???

MarcoMers, 2019-04-11:
Anything new...??? I have started the node but it seems that doesn't work properly... i have to make more test

virraaj, 2019-05-03:
Hi, Sorry for the late reply.  Earlier I tried running this package in amd64 architecture with ubuntu 14.04 and ros jade. But after connecting the same package to Raspberry Pi with armhf architecture ubuntu 18.04 and ros melodic, It is atleast publishing the topics. But still, I do have received those NACK error msgs.

huhaohao, 2019-07-04:
Hi virraaj,
i try to also use the C099-F9P evaluation board (with ZED-F9P hardware) on ubuntu 18.04 bionic and ros melodic. You said you can run this package on this environment and receive messages and ERROR messages. I want to ask, if the received messages are all okay for use or there are some error in it.

virraaj, 2019-07-04:
Well, that depends on what kind of messages are you looking for. For example, the satellites messages can be verified with the location and the UTC constant which seem to be accurate.  I think those NACK messages are because of this latest protocol which is not yet added in this package.

huhaohao, 2019-07-04:
Hi virraaj,

thanks for the information. I will buy one board and have a try.