KumarRobotics/ublox #56: How to fix this error: U-blox: received NACK: 0x06 /0x01

QuinnZhuu, 2019-07-18:
I have one GPS receiver with U-BLOX M8030 72CH , but when I run the launch file. This error occurred.
[image]

I wonder how to solve this problem
Thanks!

CK-Sung, 2020-01-04:
Have you solved this problem? I have same error "received NACK: 0x06 0x01".

SilvainLouis, 2020-03-13:
Same probleme when I try to enable other than GPS system

JihunDev, 2020-06-04:
I am developing using ROS2.

The error "U-blox: received NACK: 0x06 /0x01"  occurs.

but check it using ROS2 Topic, data comes in.

If you have solved this problem, please share it.

NovoG93, 2020-08-17:
Anyone has an update on this?

vilktor370, 2021-07-27:
Yeah i can get data but sometimes to data is 0.0 in both latitude and longitude. So I am not sure what is going on here :(

[image]

trayveet, 2022-05-12:
I had similar issues. I was able to trace it the USB HUB I was using. Using a different hub fixed that problem, but other issues arised as well. 
Hope it helps

tomy983, 2022-11-06:
```
publish:
    aid/all: false
```
instead of the incorrect 
```
publish:
    aid: false
```
fixed it for me with ZED F9P. Find if you are trying to publish incorrect messages...

gokulp01, 2023-07-10:
Has anyone fixed this issue? This is persistent in both ros2 foxy and ros2 galactic. Any idea how to fix it?

Help will be much appreciated, thanks!

gokulp01, 2023-07-13:
Update: I fixed the issue. The issue is in the config file. Make sure to use the right config file. (I also updated the firmware through the U-blox center, but I don't think that was very helpful)