KumarRobotics/ublox #60: Issue changing frequency

ChrisGermany, 2019-08-08:
I am not able to change the GPS and ROS update rate. I am using an f9p and would like to get data at 10 hz. 

My configuration file looks as follows

```# Configuration Settings for ZED-F9P device

debug: 1                    # Range 0-4 (0 means no debug statements will print)

device: /dev/ubloxF9P
frame_id: gps
rate: 10
nav_rate: 10

fix_mode: auto

uart1:
  in: 0
  out: 0

usb:
  in: 32                     # RTCM 3
  out: 0                    # No UART out for rover

# Enable u-blox message publishers
publish:
  all: false
  nav:
    all: true
    relposned: true
    posllh: true
    posecef: true
```

launching the node gives me the following output 

```[ INFO] [1565274373.379511412]: U-Blox: Opened serial port /dev/ubloxF9P
[DEBUG] [1565274373.880197059]: U-Blox: Set ASIO baudrate to 9600
[DEBUG] [1565274373.880309379]: Configuring UART1 baud rate: 9600, In/Out Protocol: 0 / 0
[DEBUG] [1565274373.882758226]: EXT CORE 1.00 (94e56e), HW VER: 00190000
[DEBUG] [1565274373.882857437]: ROM BASE 0x118B2060
[DEBUG] [1565274373.882897782]: FWVER=HPG 1.11
[DEBUG] [1565274373.882926557]: PROTVER=27.10
[DEBUG] [1565274373.882953912]: MOD=ZED-F9P
[DEBUG] [1565274373.882994036]: GPS;GLO;GAL;BDS
[DEBUG] [1565274373.883023069]: QZSS
[ INFO] [1565274373.883112140]: U-Blox Firmware Version: 8
[ INFO] [1565274373.890233711]: U-Blox configured successfully.
[DEBUG] [1565274373.890339632]: Subscribing to U-Blox messages
[DEBUG] [1565274373.922967659]: Configuring INF messages
```
Everything starts, but I am not able to get the messages at any other speed than 1Hz.

closed 2019-08-08 by ChrisGermany

Chuan701, 2020-05-14:
How to solve the problem? I have a similar problem on ROSㄡ

Chuan701, 2020-05-14:
How to solve the problem? I have a similar problem on ROS.

ChrisGermany, 2020-05-14:
I am sorry there is no real fix in the ros package I could get to work. It seems like the configuration via this package does not work for the ublox module. The way I did it is changing it in u-center and saving the configuration to the device.

Chuan701, 2020-05-14:
Thank you for your prompt response.
Could I consult you about how to save the configuration to the device in u-center?

ChrisGermany, 2020-05-14:
Yes I can do that. I think ublox forum is a good place to start. However if you have issues just let me know.

Chuan701, 2020-05-18:
Thank you for your reply.
[image]
[image]

The coordinate is WGS-84(Latitude、Longitude、Altitude) above.
But I have the problem to consult you.
In order to fuse the location data, I want to transfer the WGS-84 into ECEF or Body frame.
What could I do?

Best regards

ChrisGermany, 2020-05-21:
Are you using the F9P with a reference basestation station for rtk corrected gps?

Chuan701, 2020-05-21:
I can transfer the WGS-84 into ECEF frame successfully now.
Do you have any method or idea to transfer the ECEF to ENU(east, north, up)??

My object is to achieve the autonomous navigation.
(ps: I used the ZED-F9P with a reference base station station for rtk corrected gps)
Thank you for your help.

ChrisGermany, 2020-05-21:
If you use the F9P with a basestation you should configure the GPS in u-center to publish RELPOSNED. This is the rover position relative to base. You could take your initial reading as an offset to get the origin of the rover. After that it is a just a matter of moving NED to ENU, which does not need a complicated conversion. The message you should see in this package is [here.](https://github.com/KumarRobotics/ublox/blob/master/ublox_msgs/msg/NavRELPOSNED.msg)

Chuan701, 2020-05-22:
OK, thank you for your suggestion.  $roslaunch ublox_gps ublox_device.launch
[image]
[image]
in node.h

My topic /ublox_gps/NED_position has no information.
Could you give me some advice?

Thank you!

ChrisGermany, 2020-05-22:
Ok so looking at your code it seems like you are just creating an empty message and publishing it. That is not what I meant.
The way to go is:
1. enable NAV-RELPOSNED in u-center and safe configuration to ZED F9P
2. change your cfg in ros for the ublox by adding
```
publish:
  nav:
    relposned: true
```
3. launch and check if the ublox package is publishing a relposned message

Maybe also check out this package https://github.com/ros-agriculture/ublox_f9p
this might already do what you need

Chuan701, 2020-05-22:
I can read iTow info in /ublox_gps/navrelposned(topic), but relPosN、 relPosE、 relPosD  doesn't show the value. Could you tell me what should I do? 
[image]

Thank you!