KumarRobotics/ublox #61: How to use this with EVK M8N?

chandailrc, 2019-08-11:
Hello 

I have been trying to figure out what would change in the config file, but in vain. Can anyone point me in the right direction? The readme suggests to look at the documentation that came with the particular evk to modify the config files. I can't seem to find/understand anything for EVK- M8N. 

Thanks!

chandailrc, 2019-08-11:
Just an update. I have been able to run it with the m8u_rover.yaml config file. However, I get a lot uncertainty in the data even though the device/antenna are mostly stationary. The covariance matrix attached with the /fix message starts off with high values of around 1000 and then drops down to somewhere around 4-10. However, it still does not remains stable, and is very erratic. 
I am still not sure whether the m8u_rover.yaml config fits for the EVK-M8N.

Any help would be appreciated.

G-Pereira, 2020-03-23:
Hello, I am not able to run this for NEO-M8N, have you made any edits to `m8u_rover.yaml`? Besides the USB port of course.