KumarRobotics/ublox #63: NavSat message flags means? Help!

MarcoMers, 2019-09-06:
Hello everyone, recently i have logged a bag file which contain navsat topic, the topic have this structure: http://docs.ros.org/kinetic/api/ublox_msgs/html/msg/NavSAT_SV.html

**My question is: How to read and what does it mean the flags message???**

Here i attach a pic :
[image]

Thank you for the help!

versatran01, 2019-09-06:
flags is a bitmask as shown in the link you attached, you can decode it by some bitwise operations.

MarcoMers, 2019-09-06:
Thank you, could you send me the link of  the file where is explained how to decode the bitmask or explain how to do it?
Thank you for the  help.

versatran01, 2019-09-06:
You already have it, which is the link you attached. 
You can google bitmask for more info.

closed 2019-09-10 by versatran01