KumarRobotics/ublox #65: debian release

Timple, 2019-11-19:
With your acceptance of PR #64 your package seems quite useful to us. Would you be interested to create a debian release for the ROS repositories?
http://wiki.ros.org/ROS/ReleasingAPackage

If you don't have the time or interest available for this, I could do it for you. That would mean a new repository would need to be created [ublox-release](https://github.com/KumarRobotics/ublox-release) for which I would need write access.

Thanks in advance for the consideration.

versatran01, 2019-11-19:
Thanks for the suggestion and I really appreciate it.
I will create a new repo and add you.

Timple, 2019-11-19:
Just let me know when I have access to the release repository :+1:

Timple, 2019-11-19:
Great, I already had that I see.
For adding tags I would also need write access to this repository. (I cannot add these to a PR)

If you don't feel comfortable, feel free to add the tags yourself. I took the liberty of also tagging past releases. 
Just perform the following list of commands:

```
roscd ublox_gps
git pull
git tag 0.0.0 fb893339efe8e2c2b239c3405c337f879667c11f
git tag 0.0.1 fc931649a9e88c623c611c8d5f24164245427d1b
git tag 0.0.2 5c438441d5cb08e8c2beb9cce5f25004a359919a
git tag 0.0.3 c2d561ee21ee4435987037348e13a601fbdb86b1
git tag 0.0.4 e2d29c51bcd33e3f8c7c971c6e1c85493d524a4a
git tag 0.0.5 f9806555d4d0b5f0b9cb748ef5bd86895399923d
git tag 1.0.0 c8bb327f30fc27eb84c64e022cc73497a75306f6
git tag 1.1.0 7aa1bc5ecc206e085ef57a0b89445ee2861980c4
git tag 1.1.2 b84276abf113361bb3fae5690989d7dd8132940a
git tag 1.2.0 9a657c103ae2184e0f4c501d467e795ccd1b8f57
git push --tags
```

Tagging the verions is just a convenience, we can also do without when releasing: https://github.com/ros/rosdistro/pull/23085

versatran01, 2019-11-19:
@kartikmohta @tdinesh @ke-sun 
Can one of you add @Timple as a maintainer or give him write access to this repo?

kartikmohta, 2019-11-20:
Looks like the write access is already done.

@Timple Thanks for putting in the effort to release this.

closed 2019-11-20 by kartikmohta

Timple, 2019-11-20:
Actually the write access to this repo has not been done. I only have access to the release repository.
So would you be so kind to push the tags as described above or give me access to do this in the future?

versatran01, 2019-11-20:
@Timple 
I ran the commands to add the tags.

@kartikmohta 
I only added him to the ublox-release, not this repo.

kartikmohta, 2019-11-21:
Oops sorry I misunderstood that.

@Timple gave you write access to this repository as well now.