KumarRobotics/ublox #77: uart1:baudrate cannot be configured on f9p

Timple, 2020-02-17:
Also holds for:
```yaml
uart1:
  baudrate:
  in:
  out:
uart2:
  baudrate:
  in:
  out:
gnss:
  gps: 
  glonass: 
  beidou: 
```

USB baudrate still works though.

Seems like the interface description has changed more than expected / implemented in #64 
The list above is what we encountered and is not a full list.

Basic operation works though including the imu heading message.

0ldWang, 2020-02-19:
what's the error message when using uart?

Timple, 2020-02-19:
We got TX-errors because the buffers get full. We can verify via u-center that the baudrates are not being set to the values we pass to `ublox_gps`

0ldWang, 2020-02-19:
I also encountered a similar issue about the uart1 when using RS232 cable.
Sometimes it reported “Could not configure serial baud rate”, and sometimes reported "txbuf alloc" error.
Did you find a solution about this?

Timple, 2020-02-19:
Yes, simply put a higher baudrate using u-center.
But it would be nice if everything could be configured using this package.

0ldWang, 2020-02-20:
@Timple 
Glad you solve this, BTW, what baud rate did you set in U-Center when you use RS232 Uart Serial Cable?

MCFurry, 2020-03-02:
@0ldWang currently we use a (high) baudrate of 460800, but this largely depends on your hardware capabilities and what type of data at what rate you want to send.

On-topic: We noticed that the f9p uses quite a couple of new configuration messages. Also for setting baudrates it is using new message types. In u-center you can see them in the "Generation 9 Configuration View".

So this is at least why setting the baudrate via the old messages is not supported anymore on the f9p, although some configuration settings CAN still be applied.

I'm not sure whether I can spend some time in the near future on implementing the new configuration messages, but this should contribute to better f9p support and future chips.

0ldWang, 2020-03-03:
@MCFurry 
Thanks for your information, it's really helpful.

scottnothing, 2020-03-28:
@MCFurry According to the manual, the old message type is deprecated, but is still functional. I have been using this repository (and some changes) to configure the F9P module on the Ardusimple board, which interfaces using USB (uart1 currently unused).
I' happy to provide details on getting the ardusimple system running, which may of course diverge for others' implementation the F9P.

Chuan701, 2020-05-14:
I use the code
rosrun ublox_gps ublox_gps.(or roslaunch ublox_gps ublox_device.launch)
[image]

How can I do to run the code successfully?         Thank you very much!