KumarRobotics/ublox #79: Error : U-blox: received NACK

WbHappy, 2020-02-24:
Hello,

Thanks for the package I have been using it for quite some time with an EVK-7P.

I recently bought a EVK-M8U and I have been struggling with errors that makes this development kit unusable... Here is the print of the error when debug parameter is set at 1 :

```bash
process[ublox_m8u-2]: started with pid [7169]
[DEBUG] [1582555174.866217214]: device name: /dev/ttyACM0
[ INFO] [1582555174.875337094]: U-Blox: Opened serial port /dev/ttyACM0
[DEBUG] [1582555174.875479295]: Configuring UART1 baud rate: 115200, In/Out Protocol: 7 / 7
[DEBUG] [1582555174.877898558]: EXT CORE 3.01 (1ec93f), HW VER: 00080000
[DEBUG] [1582555174.877912486]: ROM BASE 2.01 (75331)
[DEBUG] [1582555174.877928192]: FWVER=UDR 1.21
[DEBUG] [1582555174.877935317]: PROTVER=19.20
[DEBUG] [1582555174.877942318]: MOD=NEO-M8U-0
[DEBUG] [1582555174.877949081]: FIS=0xEF4015 (84524)
[DEBUG] [1582555174.877956001]: GPS;GLO;GAL;BDS
[DEBUG] [1582555174.877962518]: SBAS;IMES;QZSS
[ INFO] [1582555174.877982330]: U-Blox Firmware Version: 8
[DEBUG] [1582555174.889008132]: Configuring measurement rate to 1000 and nav rate to 1
[DEBUG] [1582555174.889582592]: Configuring SBAS: usage 0, max_sbas 0
[DEBUG] [1582555174.890572737]: Disabling PPP
[ERROR] [1582555174.891307805]: U-blox: received NACK: 0x06 / 0x23
[FATAL] [1582555174.891406477]: Error configuring u-blox: Failed to disable PPP.
[ERROR] [1582555174.892228621]: U-Blox ASIO input buffer read error: Operation canceled, 0
[ INFO] [1582555174.892311468]: Closed connection to /dev/ttyACM0.
[ublox_m8u-2] process has finished cleanly
```

I precise that this EVK M8U works fine on Windows's U-Center.

Before posting this issue I looked a bit at how to fix and here is also my dirty workaround.
Error displayed previously is triggered because of disabling PPP functionality. Disabling PPP is done by the method `setPpp` line 454 of src/node.cpp, I commented the full block (liens 454 to 457).. 

Next error is : 
``` bash
process[ublox_m8u-2]: started with pid [9340]
[DEBUG] [1582555696.275093408]: device name: /dev/ttyACM0
[ INFO] [1582555696.283741965]: U-Blox: Opened serial port /dev/ttyACM0
[DEBUG] [1582555696.283917665]: Configuring UART1 baud rate: 115200, In/Out Protocol: 7 / 7
[DEBUG] [1582555696.287124105]: EXT CORE 3.01 (1ec93f), HW VER: 00080000
[DEBUG] [1582555696.287140639]: ROM BASE 2.01 (75331)
[DEBUG] [1582555696.287159165]: FWVER=UDR 1.21
[DEBUG] [1582555696.287168210]: PROTVER=19.20
[DEBUG] [1582555696.287177422]: MOD=NEO-M8U-0
[DEBUG] [1582555696.287187011]: FIS=0xEF4015 (84524)
[DEBUG] [1582555696.287198243]: GPS;GLO;GAL;BDS
[DEBUG] [1582555696.287207997]: SBAS;IMES;QZSS
[ INFO] [1582555696.287233075]: U-Blox Firmware Version: 8
[DEBUG] [1582555696.299533907]: Configuring measurement rate to 1000 and nav rate to 1
[DEBUG] [1582555696.300113809]: Configuring SBAS: usage 0, max_sbas 0
[DEBUG] [1582555696.300797860]: Setting wknRollover to 0
[ERROR] [1582555696.301206475]: U-blox: received NACK: 0x06 / 0x23
[FATAL] [1582555696.301285434]: Error configuring u-blox: Failed to set wknRollover: 0.
[ERROR] [1582555696.302097083]: U-Blox ASIO input buffer read error: Operation canceled, 0
[ INFO] [1582555696.302170844]: Closed connection to /dev/ttyACM0.
[ublox_m8u-2] process has finished cleanly
```

Same method, I commented the method `setWknRollover` from line 458 to 463 in src/node.cpp.

Next error is : 
``` bash
process[ublox_m8u-2]: started with pid [10666]
[DEBUG] [1582555802.100682633]: device name: /dev/ttyACM0
[ INFO] [1582555802.108730291]: U-Blox: Opened serial port /dev/ttyACM0
[DEBUG] [1582555802.108865381]: Configuring UART1 baud rate: 115200, In/Out Protocol: 7 / 7
[DEBUG] [1582555802.111237487]: EXT CORE 3.01 (1ec93f), HW VER: 00080000
[DEBUG] [1582555802.111250233]: ROM BASE 2.01 (75331)
[DEBUG] [1582555802.111262677]: FWVER=UDR 1.21
[DEBUG] [1582555802.111270182]: PROTVER=19.20
[DEBUG] [1582555802.111278119]: MOD=NEO-M8U-0
[DEBUG] [1582555802.111284640]: FIS=0xEF4015 (84524)
[DEBUG] [1582555802.111291348]: GPS;GLO;GAL;BDS
[DEBUG] [1582555802.111297726]: SBAS;IMES;QZSS
[ INFO] [1582555802.111317015]: U-Blox Firmware Version: 8
[DEBUG] [1582555802.122549721]: Configuring measurement rate to 1000 and nav rate to 1
[DEBUG] [1582555802.123297852]: Configuring SBAS: usage 0, max_sbas 0
[DEBUG] [1582555802.124136638]: Setting dynamic model to 7
[DEBUG] [1582555802.124704761]: Setting fix mode to 2
[DEBUG] [1582555802.125776732]: Setting DR Limit to 0
[DEBUG] [1582555802.127581598]: Setting time to UTC time
[DEBUG] [1582555802.128381974]: TIM-TM2 send rate on current port set to 1
[DEBUG] [1582555802.129447678]: Read GNSS config.
[DEBUG] [1582555802.129466770]: Num. tracking channels in hardware: 28
[DEBUG] [1582555802.129474526]: Num. tracking channels to use: 28
[DEBUG] [1582555802.129481288]: U-Blox GNSS configuration is correct. GNSS not re-configured.
[DEBUG] [1582555802.129492920]: Time Pulse frequency set to 2,2 on channel 1
[DEBUG] [1582555802.129500578]: msg flags set to: 0x6f
[DEBUG] [1582555802.130271126]: Enabling ADR/UDR
[ERROR] [1582555802.130733571]: U-blox: received NACK: 0x06 / 0x23
[FATAL] [1582555802.130838917]: Error configuring u-blox: Failed to enable use_adr
[ERROR] [1582555802.131759406]: U-Blox ASIO input buffer read error: Operation canceled, 0
[ INFO] [1582555802.131857651]: Closed connection to /dev/ttyACM0.
[ublox_m8u-2] process has finished cleanly
```

This time the error is regarding some enable_use_adr. I commented the block, line 1353 to 1355 in src/node.cpp that contains the `setUseAdr` method responsable for my error.

And finally I managed to use the EVK M8U.

I obviously agree that this is totally a dirty fix but only a temporary workaround. 
I would appreciate a cleaner fix in the future but at least this post might help a bit ;).

riebl, 2020-06-02:
We have observed the same issue in the past. The actual reason for this behaviour is the changed protocol by recent firmware versions of M8U. This has been fixed by #89.

WbHappy, 2020-06-02:
Thanks for the answer, I will check this.

I close the issue for now then.

closed 2020-06-02 by WbHappy

Muhammad540, 2024-11-08:
> We have observed the same issue in the past. The actual reason for this behaviour is the changed protocol by recent firmware versions of M8U. This has been fixed by #89.

Hey @riebl , i was working with the ros2 branch for the ublox-m8u module and experienced this issue of firmware version. Following the merge commit history #89, i have fixed the issue ros2 branch. If you would like to see the changes in this branch, i can open a PR for this fix.

Muhammad540, 2024-11-08:
> Hi @Muhammad540, I think you should discuss with the project maintainers if your changes should get merged. I am no longer using this ROS ublox node for various reasons.

got it, thanks for letting me know.