KumarRobotics/ublox #82: Failed to poll MonVER & set relevant settings

kevinliu257, 2020-03-26:
Hello，

When I running  by launch file, but the program hang up, the Error is  "  Failed to poll MonVER & set relevant settings" 

kevin@kevin-virtual-machine:~/catkin_ws/src/ublox-master/ublox_gps/launch$ roslaunch ublox_device.launch
... logging to /home/kevin/.ros/log/ba984710-6f31-11ea-889b-d31927527346/roslaunch-kevin-virtual-machine-87976.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/kevin/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://127.0.0.1:44973/

SUMMARY
========

CLEAR PARAMETERS
 * /ublox_gps/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ublox_gps/debug: 1
 * /ublox_gps/device: /dev/ttyUSB0
 * /ublox_gps/dgnss_mode: 3
 * /ublox_gps/dr_limit: 0
 * /ublox_gps/dynamic_model: airborne2
 * /ublox_gps/enable_ppp: False
 * /ublox_gps/fix_mode: auto
 * /ublox_gps/frame_id: gps
 * /ublox_gps/gnss/beidou: False
 * /ublox_gps/gnss/glonass: True
 * /ublox_gps/gnss/qzss: False
 * /ublox_gps/inf/all: True
 * /ublox_gps/nav_rate: 4
 * /ublox_gps/publish/aid/hui: False
 * /ublox_gps/publish/all: True
 * /ublox_gps/publish/nav/posecef: False
 * /ublox_gps/rate: 4
 * /ublox_gps/save/device: 4
 * /ublox_gps/save/mask: 3103
 * /ublox_gps/uart1/baudrate: 38400
 * /ublox_gps/uart1/in: 32
 * /ublox_gps/uart1/out: 0

NODES
  /
    ublox_gps (ublox_gps/ublox_gps)

auto-starting new master
process[master]: started with pid [87986]
ROS_MASTER_URI=http://127.0.0.1:11311

setting /run_id to ba984710-6f31-11ea-889b-d31927527346
process[rosout-1]: started with pid [87999]
started core service [/rosout]
process[ublox_gps-2]: started with pid [88016]
[ INFO] [1585207000.055439752]: U-Blox: Opened serial port /dev/ttyUSB0
[DEBUG] [1585207000.558687442]: U-Blox: Set ASIO baudrate to 9600
[DEBUG] [1585207001.065283482]: U-Blox: Set ASIO baudrate to 19200
[DEBUG] [1585207001.572044700]: U-Blox: Set ASIO baudrate to 38400
[DEBUG] [1585207001.572119662]: Configuring UART1 baud rate: 38400, In/Out Protocol: 32 / 0
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to poll MonVER & set relevant settings

kevinliu257, 2020-03-26:
I find the reason is "callbacks_.read(message, timeout);" always return false.
but I don't know how to fix it.   my OS is Ubuntu 16.04.

code as below:
template <typename T>
bool Gps::read(T& message, const boost::posix_time::time_duration& timeout) {
  if (!worker_) return false;
  return callbacks_.read(message, timeout);
}

clint-qurrent, 2021-05-13:
I'm having the exact same problem. I do have the ZED-F9P connected to u-center via USB and am attempting to read out the GPS data via UART1.

Any help on this issue would be greatly appreciated.

arpaterson, 2022-10-18:
Also experiencing this. In general this node is quite difficult to use with F9P, even with `config_on_startup: false` and manual configuration in u-center.

Frequent issue on startup:
```
[ INFO] [1666109469.028044469]: U-Blox: Opened serial port /dev/ttyACM0
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to poll MonVER & set relevant settings
```
Also, after a successful startup and shutdown of the node, it rarely connects successfully a second time.