KumarRobotics/ublox #85: Problem building on raspberry pi 4

Chrismmint, 2020-05-03:
I get the following error when building on a pi4 with ROS melodic -

[ 98%] Linking CXX executable /home/pi/catkin_ws/devel/lib/ublox_gps/ublox_gps
/usr/bin/ld: CMakeFiles/ublox_gps_node.dir/src/node.cpp.o: in function `ublox_node::UbloxNode::configureUblox()':
node.cpp:(.text+0x8aac): undefined reference to `ublox_node::UbloxNode::kResetWait'
/usr/bin/ld: /home/pi/catkin_ws/devel/lib/libublox_gps.so: undefined reference to `ublox_gps::Gps::kSetBaudrateSleepMs'
collect2: error: ld returned 1 exit status
make[2]: *** [ublox/ublox_gps/CMakeFiles/ublox_gps_node.dir/build.make:160: /home/pi/catkin_ws/devel/lib/ublox_gps/ublox_gps] Error 1
make[1]: *** [CMakeFiles/Makefile2:4785: ublox/ublox_gps/CMakeFiles/ublox_gps_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed

Any ideas?

Also, I only have a neo-6m at the moment.  Is there a previous version of this that would be recommended?

Timple, 2020-05-08:
Might be a resource issue?
Can you try with: 
`catkin build -p1 -j1` 
This will only compile one executable at a time possibly preventing memory issues

kartikmohta, 2020-05-20:
@Chrismmint I think that issue happens when using an old compiler. What version of gcc are you using? (You can check with `gcc -v`)

MLEEKIMM, 2020-05-21:
in ublox_gps folder, change cmakelist file line 27 (may be) 
SET(CMAKE_CXX_FLAGS  "${CMAKE_CXX_FLAGS} -std=c++11 -pthread")
to 
SET(CMAKE_CXX_FLAGS  "${CMAKE_CXX_FLAGS} -std=c++17 -pthread")

bjajoh, 2020-11-10:
@MLEEKIMM Thanks, that helped me!