KumarRobotics/ublox #87: [DEBUG]  I have ROM BASE 0x118B2060 on ROS.

Chuan701, 2020-05-14:
I use the code 
rosrun ublox_gps ublox_gps.(or roslaunch ublox_gps ublox_device.launch)

[image]
[image]

How can I do to run the code successfully?

scottnothing, 2020-05-16:
Are you trying to run the device as a Rover? Add `tmode3: 0` to the yaml config file.

Chuan701, 2020-05-18:
Thanks for your reply~
I can read the information successfully on ROS. 
[image]
[image]
The coordinate is WGS-84(Latitude、Longitude、Altitude) above.
But I have the problem to consult you.
In order to fuse the location data, I want to transfer the WGS-84 into ECEF or Body frame.
What could I do?  

Best regards

Chuan701, 2020-05-21:
Hello, could you help me to convert WGS84(lat, lon, alt) to ECEF Cartesian in C++??  
Thank you very much!

scottnothing, 2020-05-25:
[Sure](https://www.google.com/search?q=convert+WGS84(lat%2C+lon%2C+alt)+to+ECEF+Cartesian+in+C%2B%2B&oq=convert+WGS84(lat%2C+lon%2C+alt)+to+ECEF+Cartesian+in+C%2B%2B&aqs=chrome..69i57&client=ubuntu&sourceid=chrome&ie=UTF-8)