KumarRobotics/ublox #90: Unable to launch ZED-F9P - asking for TMODE to be set

arunavanag591, 2020-05-14:
Hello I recently got this ZED F9P unit, and when I tried launching it I received an error as 
`roslaunch ublox_gps ublox_device.launch node_name:=ublox_gps param_file_name:=nmea`

```
ROS_MASTER_URI=http://localhost:11311

process[ublox_gps-1]: started with pid [6468]
[ INFO] [1589497175.320461884]: U-Blox: Opened serial port /dev/ttyACM0
[DEBUG] [1589497175.821338575]: U-Blox: Set ASIO baudrate to 9600
[DEBUG] [1589497175.821442989]: Configuring UART1 baud rate: 9600, In/Out Protocol: 7 / 1
[DEBUG] [1589497175.823821846]: EXT CORE 1.00 (94e56e), HW VER: 00190000
[DEBUG] [1589497175.823913201]: ROM BASE 0x118B2060
[DEBUG] [1589497175.823950931]: FWVER=HPG 1.11
[DEBUG] [1589497175.823979634]: PROTVER=27.10
[DEBUG] [1589497175.824004329]: MOD=ZED-F9P
[DEBUG] [1589497175.824040987]: GPS;GLO;GAL;BDS
[DEBUG] [1589497175.824072453]: QZSS
[ INFO] [1589497175.824136395]: U-Blox Firmware Version: 9
[ WARN] [1589497175.856387905]: For HPG Ref devices, nav_rate should be exactly 1 Hz.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Invalid settings: TMODE3 must be set
[ublox_gps-1] process has died [pid 6468, exit code -6, cmd /home/wildfly/tracking_ws/devel/lib/ublox_gps/ublox_gps __name:=ublox_gps __log:=/home/wildfly/.ros/log/a839c71c-9626-11ea-a47e-a4c3f0861258/ublox_gps-1.log].
log file: /home/wildfly/.ros/log/a839c71c-9626-11ea-a47e-a4c3f0861258/ublox_gps-1*.log
```
I am unfamiliar, how to set TMODE and what is it anyways?

Chuan701, 2020-05-21:
Maybe you can try to add tmode3: 0 to the yaml config file.

JihunDev, 2020-06-04:
The same error occurs.
 
If you solved this problem, please share it.

arunavanag591, 2020-06-04:
I just used the param file c94_m8p_base which took away the tmode error. But these param files available in repo is for a different model; my original question was find if they had a config yaml dedicated for zed f9p for the best performance. @JihunDev

F1iX, 2020-09-30:
I did not manage to find a working config for ZED-F9P rovers either, but I managed to read the correct GPS fixed from NMEA-messages using [nmea_navsat_driver](http://wiki.ros.org/nmea_navsat_driver) listening on e.g. `/dev/ttyACM0`. Hope this helps.

closed 2020-10-01 by arunavanag591