KumarRobotics/ublox #91: Prerelease for Noetic fails

Timple, 2020-05-22:
Steps to reproduce:

    sudo apt-get install python3-ros-buildfarm
    mkdir -p /tmp/prerelease_job && cd /tmp/prerelease_job

From [here](http://prerelease.ros.org/noetic) the next command is generated:

    generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml   noetic default ubuntu focal amd64   --custom-repo     ublox__custom-2:git:https://github.com/kumarRobotics/ublox:master   --level 1   --output-dir ./

Running the script gives the error:

    ./prerelease.sh 

```
[100%] Linking CXX executable /tmp/ws/devel_isolated/ublox_gps/lib/ublox_gps/ublox_gps
/usr/bin/ld: CMakeFiles/ublox_gps_node.dir/src/node.cpp.o: in function `ublox_node::UbloxNode::configureUblox()':
node.cpp:(.text+0x876f): undefined reference to `ublox_node::UbloxNode::kResetWait'
/usr/bin/ld: /tmp/ws/devel_isolated/ublox_gps/lib/libublox_gps.so: undefined reference to `ublox_gps::Gps::kSetBaudrateSleepMs'
collect2: error: ld returned 1 exit status
```

Any insights? On melodic this works.

Timple, 2020-05-24:
Update: `catkin build` also fails in 20.04 docker container. So not related to release tooling. Also docker method might be easier/faster to reproduce/fix.

G-Pereira, 2020-05-26:
I can confirm this on Debian Buster with ROS Noetic

G-Pereira, 2020-05-27:
#92 Solves this

G-Pereira, 2020-05-28:
I think this should be closed, no?

G-Pereira, 2020-05-28:
Btw, is the ROS2 version of this package usable?

Timple, 2020-05-28:
Compilation fixed and released for noetic:
https://github.com/ros/rosdistro/pull/25159

closed 2020-05-28 by Timple

clalancette, 2020-05-28:
> Btw, is the ROS2 version of this package usable?

I'm biased, since I did most of the work there, but I think so, yes.  I've been using it in one of my projects, though I've only ever tested it on a Ublox C94 M8P.