KumarRobotics/ublox #95: "U-blox: received NACK: 0x06 / 0x01" and "U-Blox ASIO input buffer read error: Operation aborted., 0 "errors when running on ROS2

JihunDev, 2020-06-04:
# System environment

> Test Board: Nvidia Xavier
> GNSS module: ZED-F9P
> Operating system: Ubuntu 18.04

# Question
I am developing on ROS2 using ZED-F9P.
Since there is no yaml file for ZED-F9P, it was created and applied as follows.

When executing, "U-blox: received NACK: 0x06 / 0x01" and "U-Blox ASIO input buffer read error: Operation aborted., 0 "errors occur.

I want to know how to solve it.

## yaml File
```
# Configuration Settings for ZED-F9P device
ublox_gps_node:
    ros__parameters:
        debug: 4 # Range 0-4 (0 means no debug statements will print)
        
        device: /dev/ttyACM0
        rate: 1
        nav_rate: 4
  
       frame_id: gnss
 
       dynamic_model: automotive
 
       tmode3: 0
       fix_mode: auto
 
       uart1:
           baudrate: 115200
           in: 35
           out: 2
 
      # Enalble GNSS
      gnss:
          gps: true
          glonass: true
          beiduo: true
          galileo: true
          qzss: true
              qzss_sig_cfg: 0x110000
 
      # Enable u-blox message publishers
      publish:
          all: false
          nav:
              all: true
              relposned: true
              posllh: true
              posecef: true
```

# Error Code
```
nvidia@tegra-ubuntu:~/adasonedrive/ros2$ ros2 launch ublox_gps ublox_gps_node.launch.py
[INFO] [launch]: All log files can be found below /home/nvidia/.ros/log/2020-06-04-15-30-09-951925-tegra-ubuntu-19624
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ublox_gps_node-1]: process started with pid [19681]
[ublox_gps_node-1] [INFO] [ublox_gps_node]: U-Blox: Opened serial port /dev/ttyACM0
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Configuring UART1 baud rate: 19200, In/Out Protocol: 32/0
[ublox_gps_node-1] [INFO] [ublox_gps_node]: EXT CORE 1.00 (61b2dd), HW VER: 00190000
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: ROM BASE 0x118B2060
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: FWVER=HPG 1.12
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: PROTVER=27.11
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: MOD=ZED-F9P
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: GPS;GLO;GAL;BDS
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: QZSS
[ublox_gps_node-1] [INFO] [ublox_gps_node]: U-Blox Firmware Version: 9
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Configuring measurement rate to 1000 ms and nav rate to 1 cycles
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Disabling PPP
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Setting dynamic model to 0
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Setting fix mode to 3
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Setting DR Limit to 0
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Read GNSS config.
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Num. tracking channels in hardware: 60
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Num. tracking channels to use: 60
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: QZSS Configuration is different 1, 1
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: QZSS Configuration: 286326785
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: QZSS Configuration: 286326785
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Re-configuring GNSS.
[ublox_gps_node-1] [WARN] [ublox_gps_node]: GNSS re-configured, cold resetting device.
[ublox_gps_node-1] [WARN] [ublox_gps_node]: Resetting u-blox. If device address changes, node must be relaunched.
[ublox_gps_node-1] [ERROR] [ublox_gps_node]: U-Blox ASIO input buffer read error: Operation aborted., 0
[ublox_gps_node-1] [INFO] [ublox_gps_node]: U-Blox: Reset serial port /dev/ttyACM0
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Disabling TMODE3
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Saving the u-blox configuration, mask 3103, device 4
[ublox_gps_node-1] [INFO] [ublox_gps_node]: U-Blox configured successfully.
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Subscribing to U-Blox messages
[ublox_gps_node-1] [ERROR] [ublox_gps_node]: U-blox: received NACK: 0x06 / 0x01
[ublox_gps_node-1] [ERROR] [ublox_gps_node]: U-blox: received NACK: 0x06 / 0x01
[ublox_gps_node-1] [DEBUG] [ublox_gps_node]: Configuring INF messages
```

closed 2020-07-29 by JihunDev

sankhakarfa, 2021-04-12:
@JihunDev  Can you elaborate how did you fix this issue?

samiamlabs, 2022-04-19:
Any chance you can share the working config here @JihunDev?
I think I have the same issue using ZED-F9P with ros2.

pdebitetto, 2022-08-16:
Hi all, I am having the same issue, using the Kumar ROS driver with an F9P.  @JihunDev , it would be great if you could point us to a more detailed description of the solution.  Very much appreciated.

lidarmansiwon, 2023-06-27:
The same problem has not yet been resolved. Let me know if you've solved it

JihunDev, 2023-06-28:
@lidarmansiwon It's been too long since I've been working on that project, and I don't have a ZED-F9P. 
If my memory is correct, the problem was solved by modifying the Config file and the code of M8P.
sorry for not being helpful

lidarmansiwon, 2023-06-28:
> I am using zed_f9p. So I try to run it after replacing it from the launch file to the zed_f9p.yaml file, but I get an error Don't you remember what you edited here? Please help me I beg you.

[image]
[image]

JihunDev, 2023-06-28:
@lidarmansiwon What comes to mind is that only glonass was used, all publish was set to false, and only 'pvt' was used. Other than that, there's nothing else that comes to mind. Sorry for not being helpful.

Franklein20, 2023-08-30:
Hello guys
Any update to fix this issue?