KumarRobotics/ublox #100: how to receive gps data faster?

uttnuji, 2020-07-22:
Currently I'm using zed-f9p with ublox and ntrip.
below is yaml file.

debug: 1                   

save:
  mask: 3103                # Save I/O, Message, INF Message, Nav, Receiver 
                            # Manager, Antenna, and Logging Configuration
  device: 4                 # Save to EEPROM

device: /dev/ttyACM0
frame_id: gps
rate: 8                     # in Hz
nav_rate: 8                 # [# of measurement cycles], recommended 1 Hz, may 
                            # be either 5 Hz (Dual constellation) or 
                            # 8 Hz (GPS only)
dynamic_model: automotive    # Airborne < 2G, 2D fix not supported (3D only), 
                            # Max Alt: 50km
                            # Max Horizontal Velocity: 250 m/s, 
                            # Max Vertical Velocity: 100 m/s
fix_mode: auto
enable_ppp: false           # Not supported by C94-M8P
dr_limit: 0

uart1:
  baudrate: 38400           # zed-f9p specific
  in: 32                    # RTCM 3
  out: 0                    # No UART out for rover

gnss:
  gps: true
  glonass: true           
  beidou: false            
  qzss: false            

dgnss_mode: 3               # Fixed mode

tmode3: 1

sv_in/min_dur: 0.01
sv_in/acc_lim: 5

inf: 
  all: true                   # Whether to display all INF messages in console

# Enable u-blox message publishers
publish:
  all: true
  aid:
    hui: false

  nav:
    posecef: false

As I set the rate and nav_rate to 8 and sv_in/min_dur to 0.01, I expected the gps value to be surveyed in at 8Hz.
But the best I could get was 1-2Hz. (I checked it through ros rqt)
So I tried changing values in node.h - kSubscribeRate, kNavSvInfoSubscribeRate, kPollDuration - but the hz in rqt rather went to 0.05Hz and don't come back to 1-2Hz even if I turn the values back and catkinmake.
How can I get receive gps data faster?

jfuea, 2020-08-07:
I think the ntrip mount point(configuration) is the limiter on the maximum navigation rate.
Also, how do you parse the corrections from ntrip to the rover via ros?

shac12, 2020-10-21:
Hi, 

Did you manage to get this working ? 

I am having issues getting my zed9fp working with ros. 

A usage section in the readme would be super helpful. 
If I do get it working I will try to post one up. 

Cheers,
Sharif

pieterblok, 2023-09-02:
@uttnuji @josuefuentesdev @shac12 

I'm not sure if this is still relevant for you, but I was able to get the Ublox ZED F9P faster (10 Hz) in ROS2 (Humble) with NTRIP.
Here's the YAML file that I've used:

```
# Configuration Settings for Ublox ZED F9P device
ublox_gps_node:
  ros__parameters:
    debug: 1                    # Range 0-4 (0 means no debug statements will print)
    device: /dev/ttyACM0
    frame_id: gps
    rate: 10.0
    nav_rate: 1
    dynamic_model: portable
    uart1:
      baudrate: 230400
    gnss:
      glonass: true
      beidou: true
      gps: true
      qzss: true
      galileo: true
      imes: false
    # TMODE3 Config
    tmode3: 0                   # Survey-In Mode
    sv_in:
      reset: false              # True: disables and re-enables survey-in (resets)
                                # False: Disables survey-in only if TMODE3 is
                                # disabled
      min_dur: 300                # Survey-In Minimum Duration [s]
      acc_lim: 3.0                # Survey-In Accuracy Limit [m]

    inf:
      all: true                   # Whether to display all INF messages in console
      
    publish:
      aid/all: false
```