KumarRobotics/ublox #111: Foxy Release

jilaada, 2020-10-12:
I'm wondering if there has been any work existing or planned to port this package to ROS2 Foxy?

clalancette, 2020-10-12:
For what it is worth, I've compiled this on Foxy with no problems.  I haven't yet had time to test it out, but I don't expect any major problems.

I *would* like to see this package released into Foxy as a proper ROS package, but I'm not quite sure if the maintainers are up for that.

jilaada, 2020-10-12:
Thanks for the reply, I had already ported the package to Foxy as well, but instead of maintaining our own fork of ublox it would be nice to have it merged into their repository and have it released into Foxy as a proper ROS package as you mentioned. I'm wondering what the process is for this?

clalancette, 2020-10-12:
> Thanks for the reply, I had already ported the package to Foxy as well, but instead of maintaining our own fork of ublox it would be nice to have it merged into their repository and have it released into Foxy as a proper ROS package as you mentioned. I'm wondering what the process is for this?

So the port of ublox to ROS 2 that I did last year is already merged into this repository in the `dashing-devel` branch; as far as I know, that is enough to make it work on Foxy as well.  If you have additional fixes beyond that, please open a pull request against the `dashing-devel` branch, and I'm happy to review (just make sure to tag me).

In terms of doing a release, that requires the maintainers to essentially follow the instructions at http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease .  The maintainers of this repository would have to do that, or give me write access so I can do it.

Timple, 2020-10-12:
For ROS1 I'm the one doing the bloom/debian releases. I've [asked](https://github.com/KumarRobotics/ublox/issues/65) for the access required and @kartikmohta was friendly enough to provide this access.

They might be willing to do the same for ROS2 @clalancette

versatran01, 2020-10-12:
@clalancette 
You do have write access to this repo.
Let me know what I can do to help with a release (eg. creating another release repo if necessary)

clalancette, 2020-10-12:
> You do have write access to this repo.

@versatran01 Sorry, I didn't realize!  Thanks.

> Let me know what I can do to help with a release (eg. creating another release repo if necessary)

Yeah, if you don't mind creating a release repository, that would be great.  Please make sure to give me write access to that repository as well.  After that, I can do the release into Foxy (and Rolling), as long as you don't mind me pushing some changelogs and tags to this source repository and the release repository.

versatran01, 2020-10-12:
We already have a ublox-release, do we need to create another one? If yes, what name should I use?
Yeah go ahead and push whatever is needed for the release.

clalancette, 2020-10-12:
> We already have a ublox-release, do we need to create another one? If yes, what name should I use?

In that case, we can just reuse the existing https://github.com/KumarRobotics/ublox-release then.  Can you please give me write access to it?

> Yeah go ahead and push whatever is needed for the release.

Perfect, thanks.

versatran01, 2020-10-13:
I added you to ublox-release.

jilaada, 2020-10-13:
> So the port of ublox to ROS 2 that I did last year is already merged into this repository in the dashing-devel branch; as far as I know, that is enough to make it work on Foxy as well. If you have [...]

@clalancette Should I still make a PR to this repository with my changes for Foxy? Unfortunately I couldn't compile it in Foxy with the dashing-devel branch like you mentioned. I needed to change some of the cmake to reflect the changes in cmake between dashing and foxy. Here is my work on it: https://github.com/jilaada/ublox/tree/foxy-devel-port. It is under review here to our fork https://github.com/tier4/ublox/pull/2.

clalancette, 2020-10-13:
Thanks to @versatran01 , the fixes for Foxy are now on the `foxy-devel` branch.  I'm going to go ahead and do a release into Foxy and Rolling from that branch.

clalancette, 2020-10-13:
All right, I've done the releases here https://github.com/ros/rosdistro/pull/26940 and here https://github.com/ros/rosdistro/pull/26941.

@versatran01 Note that I bumped the major version to 2.0.0 for the release into ROS 2.  That just means that you should stick to 1.x.y versions for ROS 1 releases.

With all of that said, I'll close this out.

closed 2020-10-13 by clalancette