KumarRobotics/ublox #113: Question: Is this package breakout board agnostic ?

shac12, 2020-10-15:
Hi, 

I was interesting in using this 

https://www.ardusimple.com/simplertk2b/#base_rover_configuration

breakout board instead of the usual Sparkfun one

https://www.sparkfun.com/products/15136

for the zed-f9p. Does anyone using this package know if the ardusimple board will work with this package ?

Cheers,
Sharif

shac12, 2020-10-22:
Hi,

After further research I have found this blog post that details the process in depth. 

https://msadowski.github.io/ardusimple-ros-integration/

There is indeed a seperate config file that needs to be used in order to make this work appropriately. 

Credit and thanks to Mateusz Sadowski for his super helpful work. 

Is it possible to merge in his changes ? 

Cheers,
Sharif

shac12, 2020-10-22:
@msadowski

msadowski, 2020-10-23:
@shac12 I don't think these changes would be needed to be merged as they are only a config file and a launch file change. With ROS you can usually create your own in your own package and it should work OK!

shac12, 2020-10-23:
Hi Mateusz,

Thanks for replying, I was just intending for the package to have something that worked with the zed f9p out of the box.

This is to say that when I tried to run this package with a zed f9p it would not work reliably as you mentioned in your post.

That is what I ended up doing to get this to work. I cloned your repo and ran it all as per your blog post. Thanks again for that btw.

fcladera, 2022-03-12:
To add to this discussion, we have been using other F9P boards successfully (such as the H-RTK F9P GNSS) with minimum or no changes in the configuration file.

closed 2022-03-12 by fcladera