KumarRobotics/ublox #117: RTK support

rafidrulz, 2020-12-21:
So I am able to compile the package and get it working together with my GPS and it is able to publish the 'Fix' data. However, this is just the normal GPS readings. I am unsure of how to use the RTK features. Can someone please help me on how I can utilise RTK?

Jconn, 2020-12-22:
This is a repost of https://github.com/KumarRobotics/ublox/issues/43

Timple, 2020-12-23:
Not perse.

#43 was about doing it over a ROS topic.
Doing RTK without ROS and using this driver works fine!

rafidrulz, 2020-12-24:
How do I connect with RTK on Ubuntu 18.04, cause right now I am able to use RTK on the u-center GUI, but it is on windows. How do I go about using it on Ubuntu? Any guides for me to follow will be greatly appreciated.

elgarbe, 2021-05-24:
currently the RTK corrections msgs (RTCM3) are traveling by another channel I think. I think this driver can't consume RTCM3 msgs. I have an ardusimple base station with XBee radio and a rover. RTCM msgs are sending using Xbee. I have rtk fix msg using this driver on ubuntu 18.04.