KumarRobotics/ublox #118: rover

danielashabat, 2020-12-22:
hi i'm trying to run the rover with roslaunch but the running stops because is says "tmode3" need to be set. though the rover don't need those  setting as I can see in the launch file of rover and because the base is running alogside.
is there any parameter I need to change to fix this?
thanks

kesava-mani, 2020-12-23:
Add the below parameter to the rover's config yaml file. 
`tmode3: 0`

The runtime error should vanish now.

igor-markovic, 2021-05-25:
I had the same issue and setting `tmode3: 0` solved it. From my ublox module datasheet:

> Time Mode is a receiver mode where the position of the receiver is known and fixed and only the time
is calculated using all available satellites. This mode allows for using the receiver as a stationary
base station.

closed 2022-08-02 by fcladera