KumarRobotics/ublox #119: How to Connect to RTK on Linux/ Ubuntu?

rafidrulz, 2021-01-22:
I am currently trying to use this GPS as a sensor on my rover. I understand that this driver provides me with the GPS readings, and I can connect to the RTK without ROS. Can someone provide me a guide on how I can do that?

flynneva, 2021-01-26:
id recommend checking out [rtklib](https://github.com/rtklibexplorer/RTKLIB.git)...seems to be the best free option out there

rafa-chao, 2021-01-27:
Rafidrulz, I am doing the same than you.
My actual situation is:
- Compile rtklib to use str2str AP to send ntrip correction to the zed-f9p. 
- str2str sends all the gps info to a tcp port.
- for testing purposes, gpsd daemon takes the information from this tcp port and it works fine. 
- now, I have to figure out how to configure ublox_gps to receive the info using tcp instead of serial port. The read me indicates that tcp is contemplated, but it has not been tested.

rafidrulz, 2021-02-01:
> id recommend checking out [rtklib](https://github.com/rtklibexplorer/RTKLIB.git)...seems to be the best free option out there
Hi @flynneva 
Thank you for your response! I have looked through the package, however, am still unsure how I can exactly go about using it? Is there any guide I can follow or do you have any guidance on how I can go about utilising this package?

flynneva, 2021-02-01:
did you [read through the manual?](https://github.com/rtklibexplorer/RTKLIB/blob/demo5/doc/manual_demo5.pdf)

also....just doing a quick google search came up with a few different `rtklib and ros` results:

https://github.com/MapIV/rtklib_ros_bridge