KumarRobotics/ublox #123: TCP connection to zed-f9p using ublox ros

rafa-chao, 2021-01-28:


rafa-chao, 2021-01-28:
Hi everybody.
I am using an ardusimplertkb2 hardware, with a zed-f9p chip. This chip incorporates an internal rtk engine, in charge of fixing the gps signal, if an ntrip stream is received with the necessary corrections.

The board has only one serial port, and to communicate with the gps, only one app can do it at the same time. This means, that, or I communicate to receive the gps data with no rtk correction, or I communicate to send ntrip correction, and then I cannot communicate to receive the fix gps data....

My idea is to use str2str AP from RTKLIB, to send to the gps the ntrip correction, and at the same time, str2str sends the gps information to a tcp port.

To test that this is working fine, I start gpsd and connects it to the tcp stream, and it works fine. 

The issue is, that I want to use this ublox software to provide ROS gps signal to my rover, but I have tried different configurations to receive this tcp signal, and I always receive the same error:

started roslaunch server http://macubuntu:39783/

SUMMARY
========

CLEAR PARAMETERS
 * /ublox/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /ublox/config_on_startup: False
 * /ublox/debug: 3
 * /ublox/device: tcp://localhost:1234
 * /ublox/dgnss_mode: 3
 * /ublox/dr_limit: 1
 * /ublox/dynamic_model: portable
 * /ublox/enable_ppp: True
 * /ublox/fix_mode: auto
 * /ublox/frame_id: gps
 * /ublox/gnss/beidou: True
 * /ublox/gnss/galileo: True
 * /ublox/gnss/glonass: True
 * /ublox/gnss/qzss: True
 * /ublox/inf/all: True
 * /ublox/nav_rate: 5
 * /ublox/publish/aid/hui: False
 * /ublox/publish/all: True
 * /ublox/publish/esf: True
 * /ublox/publish/nav/posecef: False
 * /ublox/tmode3: 0

NODES
  /
    ublox (ublox_gps/ublox_gps)

ROS_MASTER_URI=http://localhost:11311

process[ublox-1]: started with pid [5837]
[ WARN] [1611830346.982516756]: Warning: PPP is enabled - this is an expert setting.
[ INFO] [1611830346.988842846]: Connecting to tcp://localhost:1234 ...
[ INFO] [1611830346.989813162]: U-Blox: Connected to localhost:1234.
[DEBUG] [1611830346.990962344]: U-Blox sent 8 bytes: 
b5 62 0a 04 00 00 0e 34 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to poll MonVER & set relevant settings
[ublox-1] process has died [pid 5837, exit code -6, cmd  

I have tried different yaml specifications, always with the same error....
Does anybody has some experience with this software and receiving tcp stream?

Thanks !!!!!!!!!!!!!!!!!

GruNyv, 2022-03-05:
I currently face the same problem. Did you find a solution to this?

Vini-002, 2022-09-13:
Hi everyone!
I'm using the [C099-F9P application board](https://www.u-blox.com/en/product/c099-f9p-application-board) and facing the same problem.
I have followed the steps on section **6.3.1.2 Rover operation in Wi-Fi AP mode** of the [User Guide](https://content.u-blox.com/sites/default/files/documents/C099-F9P-AppBoard_UserGuide_UBX-18063024.pdf). Using netcat, I can confirm that the connection is working, as it is possible to receive NMEA messages.
```
[ INFO] [1663080277.798131405]: Connecting to udp://192.168.0.1:5555 ...
[ INFO] [1663080277.798896592]: U-Blox: Connected to 192.168.0.1:5555.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to poll MonVER & set relevant settings
[ublox-2] process has died [pid 2990, exit code -6, cmd /home/vinicius/catkin_ws/devel/lib/ublox_gps/ublox_gps __name:=ublox __log:=/home/vinicius/.ros/log/970ca612-3372-11ed-b88e-d08e79e02378/ublox-2.log].
log file: /home/vinicius/.ros/log/970ca612-3372-11ed-b88e-d08e79e02378/ublox-2*.log
```

Vini-002, 2022-09-14:
It just worked for me, but I can't explain why... only change in code I noticed was to use quotes on the UDP path.

```yaml
device: "udp://192.168.0.1:5555"
```

NehalNevle, 2023-12-05:
I am working with /dev/ttylACM0 I am still getting the error

jtaveau, 2024-06-25:
Anyone found the solution to this issue?

RaccoonAD, 2024-10-25:
Did anyone found it?