KumarRobotics/ublox #124: Sending RTCM messages to GPS

rafidrulz, 2021-02-22:
Hi, I have finally managed to get NTRIP running. I have a package that subscribes to the NTRIP client and I am able to publish the rtcm messages onto ROS. However, does anyone have any experience or advice on how I can send this rtcm messages to my GPS?

TobiMiller, 2022-03-23:
Easiest way is
https://github.com/ros-agriculture/ublox_f9p

combined with: https://github.com/ros-agriculture/ntrip_ros

fcladera, 2022-08-17:
After #136, we have successfully tested this feature using [https://github.com/LORD-MicroStrain/ntrip_client](https://github.com/LORD-MicroStrain/ntrip_client). Here is an example launch that is compatible with this package:

```
roslaunch ntrip_client ntrip_client.launch host:=your_host  mountpoint:=U-BLOX username:=your_username password:=your_password rtcm_message_package:=rtcm_msgs
```

closed 2022-08-17 by fcladera