KumarRobotics/ublox #127: Unable to Poll Monver issue

sankhakarfa, 2021-03-17:
Hi,

I have connected a zed f9p using UART1 with UBX and NMEA at 115200 baud rate but I am unable to use ublox ros as it fails with the following code:

``` 
[INFO] [1615994871.828988598]: U-Blox: Opened serial port /dev/ttyUSB2
[DEBUG] [1615994872.334160800]: U-Blox: Set ASIO baudrate to 9600
[DEBUG] [1615994872.335124369]: U-Blox sent 8 bytes: 
b5 62 0a 04 00 00 0e 34 
[DEBUG] [1615994873.052473632]: U-Blox received 1 bytes 
61 
[DEBUG] [1615994873.067852662]: U-Blox received 5 bytes 
13 41 5a 62 02 
.
.
.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to poll MonVER & set relevant settings
```

I am able to poll Mon ver using terminal and also using pyubx2.

Can someone help in resolving this issue?

sankhakarfa, 2021-03-23:
Update:
This issue is not happening when connecting the ZEDf9p eval kit to Laptop through USB and selecting dev/ttyACM0

sankhakarfa, 2021-03-25:
Hi,
It was a mistake from my side.
I needed to set the proper UART1 BAUDRATE in the yaml config file in config folder even though it is required by the node to listen to the port at that baudrate.
It is not required for USB connection but for UART1 it is required. Also UART1 should have UBX message in enabled.

Closing this issue.

closed 2021-03-25 by sankhakarfa

NehalNevle, 2023-12-06:
I am still facing the same issue with right baudrate set.

NehalNevle, 2023-12-06:
* /rosdistro: noetic
 * /rosversion: 1.16.0
 * /ublox/config_on_startup: False
 * /ublox/device: /dev/ttyACM1
 * /ublox/frame_id: gps
 * /ublox/uart1/baudrate: 38400
 * /ublox/uart1/in: 1

RaccoonAD, 2024-10-25:
Did you find a solution?

mamadouDembele, 2024-10-29:
save_on_shutdown : true in the yaml seems to solve the problem