KumarRobotics/ublox #130: Publishing high precision HPPOSECEF & HPPOSLLH messages

hortovanyi, 2021-05-07:
Using the following config zed_f9p.yaml but unable to see UBX NAV HPPOSLLH (High Precision Geodectic Position) & UBX NAV HPPOSECEF (High Precision Position ECEF) topics or option to enable them published as ros2 topics.

Have a separate feed into UART2 supplying RTCM3 on the ZED-F9P. Am able to confirm in u-center 21.02 that the configuration is correct.

``` yaml
# Configuration Settings for zed_f9p device

ublox_gps_node:
  ros__parameters:
    debug: 1                    # Range 0-4 (0 means no debug statements will print)

    save:
      mask: 3103                # Save I/O, Message, INF Message, Nav, Receiver
                                # Manager, Antenna, and Logging Configuration
      device: 4                 # Save to EEPROM

    device: /dev/ttyACM0
    frame_id: zed_f9p
    rate: 15.0                     # in Hz
    nav_rate: 4                 # [# of measurement cycles], recommended 1 Hz, may
                                # be either 5 Hz (Dual constellation) or
                                # 8 Hz (GPS only)
    dynamic_model: airborne2    # Airborne < 2G, 2D fix not supported (3D only),
                                # Max Alt: 50km
                                # Max Horizontal Velocity: 250 m/s,
                                # Max Vertical Velocity: 100 m/s
    fix_mode: 3D
    enable_ppp: true
    dr_limit: 1

    uart1:
      baudrate: 115200           # NEO-M8U specific
      in: 16                     # RTCM 3
      out: 16                    # No UART out for rover

    gnss:
      glonass: true              # Supported by NEO-M8U
      beidou: false              # Supported by NEO-M8U
      qzss: false                # Supported by NEO-M8U

    dgnss_mode: 3                # Fixed mode

    inf:
      all: true                  # Whether to display all INF messages in console

    # Enable u-blox message publishers
    publish:
      all: true
      esf: true
      aid:
        hui: false
      nav:
        posecef: true
        relposned: true

    tmode3: 0
    # arp:
    #   lla_flag: true
    #   position: m
    #   position_hp: 0.1 mm
    #   acc: 1.0
```

versatran01, 2021-05-07:
@hortovanyi Hi, this repo is currently not being actively developed. 
If you can do a PR I'd be happy to merge.

hortovanyi, 2021-05-07:
@versatran01 looking through the code are only NMEA records processed? ... can't see where UBX records are handled

sankhakarfa, 2021-08-30:
Hey these messages are not yet implemented, even I require them.

hortovanyi, 2021-08-31:
@sankhakarfa I wrote a new UBX Gen 9 based driver https://github.com/aussierobots/ublox_dgnss .. its now published under Galactic. If you want to sink data (to parquet and read in jupyter notebooks) and then analyze it check out https://github.com/aussierobots/ublox_ubx_arrow

sankhakarfa, 2021-08-31:
@hortovanyi Will it work for noetic?

hortovanyi, 2021-08-31:
@sankhakarfa no. ROS2 galactic

fcladera, 2021-08-31:
Would it be possible to port your upgrades into this driver @hortovanyi?

hortovanyi, 2021-09-16:
@fcladera sorry just saw this. No I'm using the newer UBX Gen 9 protocol. This driver here is using deprecated UBX messages and doesnt output the UBX messages as ROS2 messages with header timestamps, that I required. If you are have a ZED-F9P try installing the driver I've developed with 'sudo apt install ros-galactic-ublox-dgnss' ... i dont presently output a NavSatFix message as it seems a bit out of date when you are using DGNSS

kmhallen, 2021-10-19:
I added these messages to the ROS1 branch in PR #143