KumarRobotics/ublox #132: ROS2 Foxy IMU information

sisaha9, 2021-05-19:
Hi! I have a ZED F9P which I wanted to get both NavSatFix messages and IMU messages from. I see in the README there is support for this. How can I do it for Foxy?

clalancette, 2021-05-19:
I haven't tried it with a ZED F9P, but theoretically it should work.  You'll need to download the code into a workspace, and build it (there are no Debian packages available for this).

From there, you'll have to launch the node with the appropriate configuration.  A good configuration starting point for the ZED F9P is https://github.com/KumarRobotics/ublox/pull/131 .  Once you have a configuration that you think will work, you'll launch the whole thing with:

```
ros2 run ublox_gps ublox_gps_node --ros-args --params-file path/to/config
```

sisaha9, 2021-05-19:
I was reading the source code in the Foxy branch. I don't see IMU mentioned in the node.cpp . I am able to use the ZED F9P and get GPS information. Just not able to use it to get IMU

clalancette, 2021-05-19:
Ah yeah, we don't have support for the IMU, I don't think.

If you'd be willing to work on it and contribute a PR for it, I'd definitely review it.

igor-markovic, 2022-03-01:
The ZED F9P doesn't have an IMU. Are you talking about the ZED F9R?

sisaha9, 2022-04-03:
Ah my bad. The version history in the repo indicated that Zed-F9P does support publishing IMU messages. I see that could have probably been a typo

closed 2022-04-03 by sisaha9