KumarRobotics/ublox #137: NEO-M8M config

MoffKalast, 2021-08-08:
I've been trying to get the M8M GPS to work with this driver without any success so far. I always get the what() error when launching, and I'm not sure if there's something I haven't configured right in the yaml or is this sensor completely unsupported.

I assumed it may be a permissions issue, but the same thing happens even if launched by root.

```
ubuntu@boat:~$ roslaunch ublox_gps ublox_device.launch
... logging to /home/ubuntu/.ros/log/4604936c-f7df-11eb-af08-e1fcdc9233fe/roslaunch-boat-1747.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://boat.local:35771/

SUMMARY
========

CLEAR PARAMETERS
 * /ublox_device/

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.11
 * /ublox_device/debug: 3
 * /ublox_device/device: /dev/ttyAMA0
 * /ublox_device/dgnss_mode: 3
 * /ublox_device/dr_limit: 0
 * /ublox_device/dynamic_model: airborne2
 * /ublox_device/enable_ppp: False
 * /ublox_device/fix_mode: auto
 * /ublox_device/frame_id: gps
 * /ublox_device/gnss/beidou: True
 * /ublox_device/gnss/glonass: True
 * /ublox_device/gnss/qzss: True
 * /ublox_device/inf/all: True
 * /ublox_device/nav_rate: 1
 * /ublox_device/publish/aid/hui: False
 * /ublox_device/publish/all: True
 * /ublox_device/publish/nav/posecef: False
 * /ublox_device/rate: 1
 * /ublox_device/uart1/baudrate: 9600
 * /ublox_device/uart1/in: 32
 * /ublox_device/uart1/out: 0

NODES
  /
    ublox_device (ublox_gps/ublox_gps)

ROS_MASTER_URI=http://boat:11311

process[ublox_device-1]: started with pid [1762]
[ INFO] [1628416939.814538597]: U-Blox: Opened serial port /dev/ttyAMA0
[DEBUG] [1628416939.817493361]: Configuring UART1 baud rate: 9600, In/Out Protocol: 32 / 0
[DEBUG] [1628416939.817742710]: Waiting for ACK 0x06 / 0x00
[DEBUG] [1628416939.818577460]: U-Blox sent 28 bytes:
b5 62 06 00 14 00 01 00 00 00 d0 08 00 00 80 25 00 00 20 00 00 00 00 00 00 00 b8 6b
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not configure serial baud rate
[ublox_device-1] process has died [pid 1762, exit code -6, cmd /home/ubuntu/catkin_ws/devel/lib/ublox_gps/ublox_gps __name:=ublox_device __log:=/ho                       me/ubuntu/.ros/log/4604936c-f7df-11eb-af08-e1fcdc9233fe/ublox_device-1.log].
log file: /home/ubuntu/.ros/log/4604936c-f7df-11eb-af08-e1fcdc9233fe/ublox_device-1*.log
^C[ublox_device-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

I'm sure the serial port is configured right, since I can get good data with both `screen /dev/ttyAMA0 9600` and with the nmea_navsat_driver, which publishes correct time and fix, but velocity seems to be always NaN there (which I can read fine in u-center if I connect via FTDI to a windows machine, so the sensor does calculate and report it properly).

I was hoping that this official driver may be more compatible, but apparently not ha.

My current config is as follows, though I'm not sure what half of these params even do:

```
debug: 3                    # Range 0-4 (0 means no debug statements will print)

device: /dev/ttyAMA0
frame_id: gps
rate: 1                     # in Hz
nav_rate: 1                 # [# of measurement cycles], recommended 1 Hz, may 
                            # be either 5 Hz (Dual constellation) or 
                            # 8 Hz (GPS only)
dynamic_model: airborne2    # Airborne < 2G, 2D fix not supported (3D only), 
                            # Max Alt: 50km
                            # Max Horizontal Velocity: 250 m/s, 
                            # Max Vertical Velocity: 100 m/s
fix_mode: auto
enable_ppp: false           # Not supported by C94-M8P
dr_limit: 0

uart1:
  baudrate: 9600           # C94-M8P specific
  in: 32                    # RTCM 3
  out: 0                    # No UART out for rover

gnss:
  glonass: true             # Supported by C94-M8P
  beidou: true              # Supported by C94-M8P
  qzss: true                # Supported by C94-M8P

dgnss_mode: 3               # Fixed mode

inf: 
  all: true                   # Whether to display all INF messages in console

# Enable u-blox message publishers
publish:
  all: true
  aid:
    hui: false

  nav:
    posecef: false
```

MoffKalast, 2021-08-08:
Ah I've figured out when the serial line from the sensor starts printing empty lines, it's after I attempt to launch this node once. And then it messes it up until the next hard reboot. Interesting.

harald85, 2021-09-08:
HI
I have the same Problem with the NEOM8Q. With 9600 Baudrate all works fine but when I try to change to a higher baudrate like 115200 it crashes all the time.

`[ INFO] [1631110291.034581619]: U-Blox: Opened serial port /dev/ttyS1
[DEBUG] [1631110291.371429961]: U-blox: reading 100 bytes
b5 62 01 07 5c 00 58 74 06 00 df 07 0a 12 00 07 03 f0 ff ff ff ff 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 98 bd ff ff ff ff ff ff 00 9b 84 df 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 ab 3b 
[DEBUG] [1631110291.622316589]: U-blox: reading 100 bytes
b5 62 01 07 5c 00 52 75 06 00 df 07 0a 12 00 07 03 f0 ff ff ff ff 80 b2 e6 0e 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 98 bd ff ff ff ff ff ff 00 9b 84 df 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 cc 0e 
[DEBUG] [1631110291.871344059]: U-blox: reading 100 bytes
b5 62 01 07 5c 00 4c 76 06 00 df 07 0a 12 00 07 03 f0 ff ff ff ff 00 65 cd 1d 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 98 bd ff ff ff ff ff ff 00 9b 84 df 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 4e 00 00 80 a8 12 01 0f 27 00 00 e0 4a 23 00 00 00 00 00 00 00 00 00 f0 bf 
[DEBUG] [1631110292.035082877]: U-Blox: Set ASIO baudrate to 9600
[DEBUG] [1631110292.106491540]: U-blox: reading 100 bytes
b5 62 01 07 5c 00 46 77 06 00 df 07 0a 12 00 07 03 f0 ff ff ff 00 7e 60 9e e0 86 00 00 60 00 00 00 00 00 00 00 00 00 00 00 00 00 80 78 fe 9e fe fe fe fe fe 00 9e 78 80 fe 7e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 18 98 00 00 00 80 e6 18 60 00 fe 80 7e 18 00 00 00 98 98 1e 18 
[DEBUG] [1631110293.035571588]: U-Blox: Set ASIO baudrate to 19200
[DEBUG] [1631110294.036179581]: U-Blox: Set ASIO baudrate to 38400
[DEBUG] [1631110295.036618554]: U-Blox: Set ASIO baudrate to 57600
[DEBUG] [1631110296.037056979]: U-Blox: Set ASIO baudrate to 115200
[DEBUG] [1631110296.037205700]: Configuring UART1 baud rate: 115200, In/Out Protocol: 7 / 1
[DEBUG] [1631110296.037325352]: Waiting for ACK 0x06 / 0x00
[DEBUG] [1631110296.037595267]: U-Blox sent 28 bytes: 
b5 62 06 00 14 00 01 00 00 00 d0 08 00 00 00 c2 01 00 07 00 01 00 00 00 00 00 be 72 
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not configure serial baud rate
`

Mario2407, 2022-03-11:
Hi,
I am also struggling with the same error and the NEO-M8N on ROS.
Do you have a solution for that?

MoffKalast, 2022-03-11:
Well fwiw, I've switched from this driver to the generic nmea_navsat_driver which worked fine after some tuning. I think ublox is not really interested in supporting these low cost modules.