KumarRobotics/ublox #145: Galactic optional parameter error.

spikyknox, 2021-11-04:
In the galactic-devel branch, if I don't set some optional parameter the node crash with : 

```
[ublox_gps_node-1] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[ublox_gps_node-1]   what():  expected [integer_array] got [not set]
```
I think the bug have been introduced with the addition of the static type during, the get_parameter needs a default param.
  
```
  this->declare_parameter<std::vector<int64_t>>("rtcm.ids");
  this->declare_parameter<std::vector<int64_t>>("rtcm.rates");
```

clalancette, 2021-11-04:
@spikyknox I don't have my ublox hardware handy, but can you give #146 a try?  I believe it will solve the issue.

spikyknox, 2021-11-04:
I give it a try and it indeed solve the problem Thanks @clalancette !

clalancette, 2021-11-04:
Great, thanks for testing.

spikyknox, 2021-11-04:
I don't know if it's related but parameter <<nmea.version>> is missing.

PR #146 (2021-11-04): Fix parameter declaration types.
Instead of trying to force things with template arguments,
instead explicitly declare the type when declaring arguments
without a default.  This should allow the types to be properly
set while still allowing these to be run without a type explicitly
set.

Fixes #145
closed 2021-11-04 by clalancette