KumarRobotics/ublox #147: galactic Error when enabling glonass,galieo or beidou. ( Zed-f9p)

spikyknox, 2021-11-05:
If I use default ( gps only) config everything work on the galactic branch. 

zed_f9p.yaml
    gnss:
      glonass: True
When I enable glonass I get this error.

`[ublox_gps_node-1] [DEBUG] [1636075968.380589616] [ublox_gps_node]: GLONASS Configuration is different
[ublox_gps_node-1] [DEBUG] [1636075968.380761942] [ublox_gps_node]: Re-configuring GNSS.
[ublox_gps_node-1] [DEBUG] [1636075968.381049056] [ublox_gps_node]: Waiting for ACK 0x06 / 0x3e
[ublox_gps_node-1] [DEBUG] [1636075968.381402221] [ublox_gps_node]: U-Blox sent 60 bytes: 
[ublox_gps_node-1] b5 62 6 3e 34 0 0 3c 3c 6 0 8 10 0 1 0 11 11 1 3 3 0 0 0 0 1 2 a 12 0 0 0 0 21 3 2 5 0 0 0 0 11 5 0 4 0 0 0 0 15 6 8 c 0 1 0 0 11 dd 7d 
[ublox_gps_node-1] [ERROR] [1636075968.381979649] [ublox_gps_node]: U-blox: received NACK: 0x06 / 0x3e
[ublox_gps_node-1] [FATAL] [1636075968.383397172] [ublox_gps_node]: Error configuring u-blox: Failed to cold reset device after configuring GNSS`

I can enable glonass with the ros-melodic driver.
How can I enable glonass on the galactic branch?

clalancette, 2021-11-05:
Sorry, I'm honestly not sure.  When I ported this to ROS 2, I only ever tested GPS (that was all I needed at the time).  I've never dealt with GLONASS before.

The easiest thing to do may be to run the Melodic driver and figure out what sequence of commands makes GLONASS work.  Then you can try to replicate that sequence on the ROS 2 side (and provide any fixes).  I'm happy to review fixes.

spikyknox, 2021-11-05:
I forgot to set the config_on_start on melodic. Now I can the same error on melodic.

spikyknox, 2021-11-08:
Is it possible that the config part of the ublox driver is not working for Firmware 9. It look like they change the protocol.
[image]
They are now using UBX-CFG-VALSET.

clalancette, 2021-11-08:
That certainly could be.  If you make it work with UBX-CFG-VALSET, then I'd be happy to review a pull request.