KumarRobotics/ublox #149: Issues publishing ZED-F9R IMU Messages

ashtencodes, 2021-11-29:
I was trying to get ZED-F9R IMU data published in the form of a sensor_msgs/imu message. Currently, I am using a zed-f9p config file that can be found below. 

```
# Configuration Settings for zed-f9p device
ublox_gps_node:
  ros__parameters:
    debug: 0                    # Range 0-4 (0 means no debug statements will print)
    device: /dev/ttyACM0
    frame_id: gps
    uart1:
      baudrate: 38400
    # TMODE3 Config
    tmode3: 1                   # Survey-In Mode
    sv_in:
      reset: True               # True: disables and re-enables survey-in (resets)
                                # False: Disables survey-in only if TMODE3 is
                                # disabled
      min_dur: 300              # Survey-In Minimum Duration [s]
      acc_lim: 3.0              # Survey-In Accuracy Limit [m]

    inf:
      all: true                   # Whether to display all INF messages in console

    publish:
      all: true
      aid:
        hui: false
      nav:
        posecef: false 
      esf:
        meas: true
```
When I run the launch file, it seems to only publish on the default fix topics. From the ublox documentation it looks like publish/esf/meas controls toggles the IMU data stream. Is there anything I am doing wrong?

versatran01, 2021-12-08:
Is this the actual config you are using?

ashtencodes, 2021-12-08:
Yes, this is the config file being used in the launch file.

versatran01, 2021-12-08:
Can you remove the first three lines? just start with debug:0

ashtencodes, 2021-12-08:
With the first three lines removed, I get most of the nav messages, but no esf messages.
I do get an error twice:

U-blox: received NACK: 0x06 / 0x01
U-blox: received NACK: 0x06 / 0x01

versatran01, 2021-12-08:
You need to follow the same format as 
https://github.com/KumarRobotics/ublox/blob/master/ublox_gps/config/c94_m8p_base.yaml

ashtencodes, 2021-12-08:
I'm still only seeing NAV topics and the NACK errors.

versatran01, 2021-12-08:
https://github.com/KumarRobotics/ublox/blob/76ff4da9ec87f47fc8e7e2f556bc88026241c05a/ublox_gps/src/node.cpp#L1368
This is where imu is published, you can add a print and see if it is reached.

ashtencodes, 2021-12-08:
I added a print statement within the if statement, the if statement is not getting triggered.

versatran01, 2021-12-08:
What about the callback?

ashtencodes, 2021-12-08:
[image]

I added a cout statement for the callback as well but not seeing anything in the terminal.

versatran01, 2021-12-08:
What I meant is right after this line
https://github.com/KumarRobotics/ublox/blob/76ff4da9ec87f47fc8e7e2f556bc88026241c05a/ublox_gps/src/node.cpp#L1367

ashtencodes, 2021-12-08:
shall I changed enabled["esf_meas"], enabled["esf"]

to enabled["true"]?

ashtencodes, 2021-12-08:
> What I meant is right after this line
> 
> https://github.com/KumarRobotics/ublox/blob/76ff4da9ec87f47fc8e7e2f556bc88026241c05a/ublox_gps/src/node.cpp#L1367
> 
> and manually set enable to true on this line
> 
> https://github.com/KumarRobotics/ublox/blob/76ff4da9ec87f47fc8e7e2f556bc88026241c05a/ublox_gps/src/node.cpp#L1340
> 
> I had a feeling that this is due to a config issue. Just make sure your config file has the right format and the launch file is using that config.

I'm currently using the config you sent me, c94_m8p_base.yaml, is there some sort of formatting issue on this file?

versatran01, 2021-12-08:
Replace
```
nh->param("publish/esf/meas", enabled["esf_meas"], enabled["esf"]); 
```
with
```
enabled["esf_meas"] = true;
```

versatran01, 2021-12-08:
Take a look at the config folder and find the right config. I just sent that one as a referece.

ashtencodes, 2021-12-08:
[image]

Ah alright, I will look into using the zedf9p config. This still doesn't seem to be printing in the terminal.

tbroed, 2022-01-17:
do you have any updates on this topic? I am also trying to get the IMU messages from the ZED-F9R

ashtencodes, 2022-01-18:
> do you have any updates on this topic? I am also trying to get the IMU messages from the ZED-F9R

Sorry no updates as of now, I am planning on looking into the source code soon. Let me know if you get anything!

spikyknox, 2022-03-12:
I have look into the source code found that 

```
if (product_category == "HPG" && ref_rov == "REF") {
    components_.push_back(std::make_shared<HpgRefProduct>(nav_rate_, meas_rate_, updater_, rtcms_, this));
  } else if (product_category == "HPG" && ref_rov == "ROV") {
    components_.push_back(std::make_shared<HpgRovProduct>(nav_rate_, updater_, this));
  } else if (product_category == "HPG") {
    components_.push_back(std::make_shared<HpPosRecProduct>(nav_rate_, meas_rate_, frame_id_, updater_, rtcms_, this));
  } else if (product_category == "TIM") {
    components_.push_back(std::make_shared<TimProduct>(frame_id_, updater_, this));
  } else if (product_category == "ADR" ||
             product_category == "UDR") {
    components_.push_back(std::make_shared<AdrUdrProduct>(nav_rate_, meas_rate_, frame_id_, updater_, this));
```

But the ZED-F9R Firmware is FWVER=HPS 1.00. The INS/imu publication seems to be in AdrUdrProduct

Jia-le-wang, 2022-07-15:
Is anyone able to run the ROS driver for Ublox F9R? I want to get IMU and GNSS data at the same time.

andrespulido8, 2024-01-30:
> I have look into the source code found that
> 
> ```
> if (product_category == "HPG" && ref_rov == "REF") {
>     components_.push_back(std::make_shared<HpgRefProduct>(nav_rate_, meas_rate_, updater_, rtcms_, this));
>   } else if (product_category == "HPG" && ref_rov == "ROV") {
>     components_.push_back(std::make_shared<HpgRovProduct>(nav_rate_, updater_, this));
>   } else if (product_category == "HPG") {
>     components_.push_back(std::make_shared<HpPosRecProduct>(nav_rate_, meas_rate_, frame_id_, updater_, rtcms_, this));
>   } else if (product_category == "TIM") {
>     components_.push_back(std::make_shared<TimProduct>(frame_id_, updater_, this));
>   } else if (product_category == "ADR" ||
>              product_category == "UDR") {
>     components_.push_back(std::make_shared<AdrUdrProduct>(nav_rate_, meas_rate_, frame_id_, updater_, this));
> ```
> 
> But the ZED-F9R Firmware is FWVER=HPS 1.00. The INS/imu publication seems to be in AdrUdrProduct. I added ''product_category == "HPS" ||'' to AdrUdr if and it start publishing esf data.

This one worked for me

Jia-le-wang, 2024-01-31:
@andrespulido8 Good job！！