KumarRobotics/ublox #150: Ntrip client to recieve correction data

felrock, 2021-12-14:
Ḯm using a zed-f9p and i would like to receive correction data from an ntrip server, so i need a ntrip client and then apply the received correction. I found this related [pr](https://github.com/KumarRobotics/ublox/pull/136), but it seems to send rtcm and not receive it. Is there another repo that could aid here?

stan-guer, 2022-01-21:
I used this and it works:
https://github.com/ros-agriculture/ntrip_ros

fcladera, 2022-08-17:
After #136, we have successfully tested this feature using [https://github.com/LORD-MicroStrain/ntrip_client](https://github.com/LORD-MicroStrain/ntrip_client). Here is an example launch that is compatible with this package:

```
roslaunch ntrip_client ntrip_client.launch host:=your_host  mountpoint:=your_mountpoint username:=your_username password:=your_password rtcm_message_package:=rtcm_msgs
```

closed 2022-08-17 by fcladera

TobiMiller, 2022-08-18:
@fcladera but how to set the config in ublox package? 

https://github.com/KumarRobotics/ublox/blob/master/ublox_gps/src/node.cpp#L1255

The code starts an subscription but param is called "publish" ?! it is misleading. can yo provide me the right instructions for the config.yaml and on which topic i have to publish the rtcm messages? Thank you

fcladera, 2022-08-18:
Yes, that's correct. The NTRIP client publishes to `/rtcm`.

euge2838, 2023-08-24:
@Squizzy911 did you solve this issue? I have the same one.

pieterblok, 2023-09-02:
> After #136, we have successfully tested this feature using https://github.com/LORD-MicroStrain/ntrip_client. Here is an example launch that is compatible with this package:
>
> ```
> roslaunch [...]

@fcladera thanks for the update! 

I'm currently using the NTRIP client of RTKLIB (https://github.com/rtklibexplorer/RTKLIB.git), but I want to switch to the NTRIP client of LORD-MicroStrain in ROS2. Doesn't this LORD-MicroStrain option need the output specification (-out)? And how can I properly convert this RTKLIB code to the LORD-MicroStrain one?

```terminal
./str2str -in ntrip://guest:guest@160.16.134.72:80/CQ-UBLOX -out serial://ttyACM0:230400
```

This is the format: ntrip://[username]:[password]@[host]:[port]/[mountpoint]

Maybe I missed it, but there's no clear example on the LORD-MicroStrain git. Thank you!