KumarRobotics/ublox #173: Humble support

samiamlabs, 2022-07-29:
Hi!

The galactic-devel would not build for me on ROS2 Humble.

Updating to the new macro for linking to interfaces generated in the same package seems to have fixed it.

```diff
include_directories(include)
add_library(${PROJECT_NAME}_lib src/ublox_msgs.cpp)
- rosidl_target_interfaces(${PROJECT_NAME}_lib ${PROJECT_NAME} "rosidl_typesupport_cpp")
+ rosidl_get_typesupport_target(cpp_typesupport_target ${PROJECT_NAME} "rosidl_typesupport_cpp")
+ target_link_libraries(${PROJECT_NAME}_lib "${cpp_typesupport_target}") 
ament_target_dependencies(${PROJECT_NAME}_lib
  "ublox_serialization"
)
```

I have a branch with the fix here https://github.com/samiamlabs/ublox/tree/humble-devel

There is currently no `humble-devel` branch in this repo to open a PR to though.

fcladera, 2022-08-02:
Just created the branch. Could you please submit your PR?

samiamlabs, 2022-08-08:
Sorry about the slow response, was on vacation last week.
PR is up here now: https://github.com/KumarRobotics/ublox/pull/178

clalancette, 2022-09-07:
This is already fixed on the `ros2` branch, so no changes should be needed.  Closing this out, but feel free to reopen if you disagree.

closed 2022-09-07 by clalancette