KumarRobotics/ublox #177: how to convert NMEA message into   xy cordinates and position ?

GaneshGovind1, 2022-08-02:
GNVTG,,T,,M,0.065,N,0.121,K,A*3C
$GNGGA,142821.00,4628.60678,N,01119.68747,E,1,08,1.47,275.0,M,46.0,M,,*43
$GNGSA,A,3,01,04,21,09,19,03,,,,,,,2.51,1.47,2.03,1*00
$GNGSA,A,3,87,72,,,,,,,,,,,2.51,1.47,2.03,2*0D
$GNGSA,A,3,,,,,,,,,,,,,2.51,1.47,2.03,3*06
$GNGSA,A,3,,,,,,,,,,,,,2.51,1.47,2.03,4*01
$GPGSV,2,1,08,01,40,147,33,03,67,061,25,04,75,183,34,09,40,219,28,1*6C
$GPGSV,2,2,08,17,35,251,27,19,35,283,22,21,18,147,29,31,23,053,15,1*6A
$GPGSV,1,1,03,01,40,147,24,04,75,183,25,09,40,219,23,6*5C
$GLGSV,1,1,03,71,27,050,26,72,71,359,31,87,67,220,32,1*48
$GLGSV,1,1,04,71,27,050,20,72,71,359,23,,,,18,,,,21,3*7D
$GAGSV,1,1,01,10,,,38,7*78
$GBGSV,1,1,00,0*77
$GNGLL,4628.60678,N,01119.68747,E,142821.00,A,A*7E
�b;(ء"iv�>�&>Nn��@���b\ء�b\����X߈�6˳92�X������������!�46�k8�\AO!*�$GNRMC,142822.00,A,4628.60679,N,01119.68748,E,0.064,,010822,,,A,V*1B
$GNVTG,,T,,M,0.064,N,0.119,K,A*36
$GNGGA,142822.00,4628.60679,N,01119.68748,E,1,08,1.47,275.0,M,46.0,M,,*4E
$GNGSA,A,3,01,04,21,09,19,03,,,,,,,2.51,1.47,2.03,1*00
$GNGSA,A,3,87,72,,,,,,,,,,,2.51,1.47,2.03,2*0D
$GNGSA,A,3,,,,,,,,,,,,,2.51,1.47,2.03,3*06
$GNGSA,A,3,,,,,,,,,,,,,2.51,1.47,2.03,4*01
$GPGSV,2,1,07,01,40,147,33,03,67,061,25,04,75,183,34,09,40,219,28,1*63
$GPGSV,2,2,07,17,35,251,27,19,35,283,22,21,18,147,30,1*5C
$GPGSV,1,1,03,01,40,147,23,04,75,183,25,09,40,219,23,6*5B
$GPGSV,1,1,01,31,23,053,,0*51
$GLGSV,1,1,03,71,27,050,26,72,71,359,31,87,67,220,32,1*48
$GLGSV,1,1,04,71,27,050,18,72,71,359,24,87,67,220,22,,,,18,3*4C
$GAGSV,1,1,01,10,,,38,7*78
$GBGSV,1,1,00,0*77
$GNGLL,4628.60679,N,01119.68748,E,142822.00,A,A*73

fcladera, 2022-08-02:
Could you please provide details on what receiver are you using, your configurations, and how you are running the driver? We can help you better if your issues are more descriptive.

closed 2022-08-03 by fcladera

GaneshGovind1, 2022-08-03:
I'm using ARDUSIMPLE AS-STARTKIT-BASIC-L1L2-NH-02 Simple RTK2B , Can i use this for localization https://github.com/MapIV/localization_evaluation_toolkit