KumarRobotics/ublox #190: How do we publish NMEA messages?

arpaterson, 2022-10-12:
I haven't had success yet, but I'm working on this, and if it does work, a closed issue covering it off will help any others in future.

**Describe the bug**
It appears that as of this [commit](https://github.com/KumarRobotics/ublox/commit/fc9fb4f0a4b4c52b988b50fdc4f47aff20aa2f79) in master branch, ublox_gps should be able to publish nmea sentences into ros.
I have checked out this commit, run rosdep install and made sure new dependancy nmea_msgs is installed. But I can't see any NMEA output,

I am using a sparkfun zed-f9p SMA board.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout the above commit into your catkin workspace
2. `rosdep install --from-paths src --ignore-src -r -y`
3. Configure the receiver using u-center, don't disable nmea messages in u-center.
4. add this to the example zed_f9p.yaml, leaving `config_on_startup: false`,
publish:
  all: true
  nmea: true
```
5. my launch file entry for the ublox node:
``` xml
<!-- ublox_f9p node -->       
<include file="$(find ublox_gps)/launch/ublox_device.launch" if="$(arg enable_ublox)">
    <arg name="param_file_name"     value="zed_f9p" />
    <arg name="param_file_dir"      value="$(find gnss)/config" />
    <arg name="node_name"           value="ublox" />
    <arg name="output"              value="screen" />
    <arg name="respawn"             value="true" />
    <arg name="respawn_delay"       value="30" />
    <arg name="clear_params"        value="true" />
</include>
```
6. See valid NavPVT being published, but no sign of nmea.

**Expected behavior**
Expected ublox/nmea topic to appear with nmea_msgs/Sentence msgs containing NMEA strings.

**Desktop (please complete the following information):**
 - Ubuntu 20.04
 - ROS1 Noetic

**Additional context**
Jetson Orin
zed-f9p

arpaterson, 2022-10-12:
@haho12 Could you weigh in on this? 
Presumably it works, and I am doing something wrong.

haho12, 2022-10-14:
@arpaterson 
Do debug this, could you first check if the NMEA messages are actually received by the program?
In ublox_serialization/include/ublox/serialization.h would you dump unused_data_ to stdout after the for loop in line ~214 ?
unused_data_ stores all data which is not recognized as a u-blox raw message, e.g. NMEA strings.
HH

haho12, 2022-10-14:
Or more easily, setting this parameter should fump received NMEA messages as well, I guess:
debug: 4

Interrollipsis, 2022-10-15:
Somewhat beginner to Ros here, it seems like were somewhat working on the same issue. Currently, is there no simple way to have a fixed base station/rover setup with 2 f9p's using this package? A simple way to subscribe to a working RTCM topic? I'm becoming exceedingly frustrated and confused and wondering what the point even is of this package.

arpaterson, 2022-10-18:
> @arpaterson Do debug this, could you first check if the NMEA messages are actually received by the program? In ublox_serialization/include/ublox/serialization.h would you dump unused_data_ to stdout [...]

I set debug 4, but I'm not sure what I am looking for as it prints a lot of binary that doesn't look like NMEA.

Do you use an f9p? It would be helpful to confirm it is user error.

I have tried `config_on_startup: false` and `true`, But as I understand it, leaving it false just means all firmware setup needs to be done through u-center, and when it is true, not all firmware config can be done by ublox_gps due to the f9p using a newer protocol.

The `publish:` section (+ `frameid` ) configures the ublox_gps node and publication of messages into ros.

The `nmea:` section I am also unsure about. I would want to disable all filtering of GxXXX strings.

I know it produces GNGGA (as it is a GNSS receiver, not just GPS), I've observed it in u-center.

arpaterson, 2022-10-18:
Adding the following
```
      }
      else {
          unused_data_.push_back(data_[0]);
          fprintf( stdout, "\n\n \e[0;31m unused_data_: %s \e[0m \n ", unused_data_.c_str() );
      }
    }
```

resulted in
```
unused_data_: $GLGSV,<obscured>
$GLGSV,<obscured>
$GLGSV,<obscured>
$GAGSV,<obscured>
$GAGSV,<obscured>
$GAGSV,<obscured>
$GAGSV,<obscured>
$GAGSV,<obscured>
$GBGSV,<obscured>
$GBGSV,<obscured>
$GBGSV,<obscured>
$GBGSV,<obscured>
$GBGSV,<obscured>
$GBGSV,<obscured>
$GQGSV,<obscured>
$GNGLL,<obscured>
```
Notably no GxGGA messages.

But... there are nmea strings being publoshed to /ublox/nmea

haho12, 2022-10-18:
Here's the config file that I use on my f9k. Any improvement?

# Configuration Settings for M8N device

debug: 0                    # Range 0-4 (0 means no debug statements will print)
                            # 4: show raw ub and nmea msgs RX+TX
                            # 2: show raw ub and nmea msgs TX

save:
  mask: 3103                # Save I/O, Message, INF Message, Nav, Receiver 
                            # Manager, Antenna, and Logging Configuration
  device: 4                 # Save to EEPROM

device: /dev/ttyACM0
frame_id: rover
rate: 4                     # in Hz
nav_rate: 4                 # [# of measurement cycles], recommended 1 Hz, may 
                            # be either 5 Hz (Dual constellation) or 
                            # 8 Hz (GPS only)
dynamic_model: pedestrian   # Airborne < 2G, 2D fix not supported (3D only), 
                            # 0: portable, <= 310 m/s, general purpose, low position accuracy
                            # 2: stationary, 0 m/s
                            # 3: pedestrian, <=  30 m/s
                            # 4: automotive, <= 100 m/s
                            # Max Alt: 50km
                            # Max Horizontal Velocity: 250 m/s, 
                            # Max Vertical Velocity: 100 m/s
fix_mode: 3d                # 1: 2d, 2: 3d, 3: auto
enable_ppp: false

uart1:
  baudrate: 115200
  in: 16                     # RTCM 3
  out: 16                    # No UART out for rover

gnss:
  gps: true
  glonass: false
  beidou: false
  qzss: false
  #sbas: true

#sbas:
#  max: 2   # obsolete
#  usage: 7 # bit0: range, bit1: differential corrections, bit2: integrity
  
#dgnss_mode: 3               # Fixed mode

inf: 
  all: true                 # Whether to display all INF messages in console

# Enable u-blox message publishers
publish:
  nav:
      all: true
#  rxm:
#      all: true # not supported ?
  esf:
    all:  true
#  aid:
#      all: true
#  mon:
#      all: true
#    hui: false
#  nav:
#    posecef: false
  hnr:
     pvt: false # setting rate gave NAK on F9K
  nmea: true

arpaterson, 2022-10-18:
> 

Thank you!

Yes I made some progress. But only with `config_on_startup: false`.

I cannot for the life of me avoid issues at startup with it enabled.

I think I am beginning to understand the limitations of this node with the f9p.
[this issue](https://github.com/KumarRobotics/ublox/issues/59) suggests that the receiver configuration is checked, and if it differs (from whats on the receiver? or defaults?) then the config message is sent, and if necessary the device is reset.
It also suggests we want to avoid that (why? I'm not sure, but it doesn't work).
So we set the config yaml to match so that those steps are not attempted.

Right now, I am setting everything very granularly in u-center, saving to the memory, and a file, and then disabling 'config_on_startup' in the yaml file.

I can then start the node a bit more reliably, instead of borking my serial port and having to reboot every second time. It still does not start reliably a second time, though.

With current master checkedout, and the desired nmea messages enabled/disabled in u-center, plus the talkerid configured again from u-center, it is now publishing nmea messages.

I will look into using the config_on_startup a bit more later, but right now its just too flaky and I don't have time for it. Your config file will be a great help.

Thanks for your contribution.regarding nmea publishing. I will use it with ntrip_client.
Previously I was assembling and partially spoofing a GGA sentence from the published nav messages.

I will look into publishing RTCM data in the same way as you implemented NMEA pass through. I want rtcm_msgs published on rtcm_out, so I can pass it directly to another ublox_node (eliminating ntrip altogether and making an offline RTK setup).
Just off the top of your head does that sound like a good approach?
I think your

arpaterson, 2022-10-18:
BTW I assumed you are using the ros1 branch, since your contribution was there?

haho12, 2022-10-19:
The idea of this package is that no settings in u-center are done since this is barely reproducible. Instead the receiver should be configured by the ROS params and saved to EEPROM if there's any change. If any change is detected, the receiver is rebooted. I've seen the reboot issue you describe as well. But if you set the params once for permanent, then this doesn't matter much. (And yes, I'm still on ROS1 here.)

arpaterson, 2022-10-21:
> The idea of this package is that no settings in u-center are done since this is barely reproducible. Instead the receiver should be configured by the ROS params and saved to EEPROM if there's any [...]

Yes, I understand that is the goal, but I have not had any luck with it.
As it stands, at least with the f9p, using the node to do the full configuration is not exactly straight forward.
The node most often just does not launch successfully. I will have to revisit that.
My grasp of it isn't good enough at this point to make a decent contribution to the documentation.

I think we can close this regarding the publish nmea feature as not a bug. (Thank you again for that feature)

Did you have any thoughts on the rtcm_msgs/Message pass through? I would be interested in implementing that feature.

akrv, 2023-02-17:
I am using swiftnav for RTCM corrections and F9P on the rover. 
I am not able to publish the NMEA sentences out of the driver.
This seems necessary for the NTRIP client to send to the server to get RTCM corrections.
Is it possible to share the config file that worked?

if I set NMEA to true, then I have to set the version and other dependent vars that I do not find the docs to set it properly.

@arpaterson can you post your config yaml file?

arpaterson, 2023-02-18:
> I am using swiftnav for RTCM corrections and F9P on the rover. I am not able to publish the NMEA sentences out of the driver. This seems necessary for the NTRIP client to send to the server to get [...]

Make sure nmea messages are enabled in the *receiver firmware*. 
My config files uses 'config_on_startup: false', so it does not configure receiver firmware options.
I checked/enabled those messages on the appropriate receiver USB/serial port using u-center.

Then also configure the `ublox_gps` node to publish nmea_msgs/Sentence messages by
**_first making sure you have checked out a late enough commit_** for that feature,
and second by enabling nmea publishing in your config file.
the only releveant line in my config file is 
```
publish:
    all: true
    nmea: true
```

akrv, 2023-02-18:
Interesting. Only difference send to be I did the repo install and didn't pull the master branch.

I'll try that and report back

arpaterson, 2023-02-18:
Yep, well if a feature is on a git branch, but has not been released into rosdep/apt, that’ll happen. Something to keep in mind with ros packages. you can include the package as a git sub module at the specific commit.

Anthony Paterson