KumarRobotics/ublox #195: Cannot configure GNSS for ZED F9R on Ubuntu 20.04 ROS 2 Foxy ARM64

duxroq, 2022-12-20:
**Describe the bug**
Repeated attempts to start ROS 2 Foxy ublox_gps node for ZED F9R device always end in a FATAL. Whether QZSS is enabled or disabled, the QZSS configuration is always "different":
```
[ublox_gps_node-1] [DEBUG] [1671498095.347203525] [ublox_gps_node]: QZSS Configuration is different 1, 1
[ublox_gps_node-1] [DEBUG] [1671498095.347210981] [ublox_gps_node]: QZSS Configuration: 286326785
[ublox_gps_node-1] [DEBUG] [1671498095.347217797] [ublox_gps_node]: QZSS Configuration: 286326785
```
This causes the node to determine that the configuration is not correct and reconfiguration is necessary. Reconfiguration always ends in a FATAL:
```
[ublox_gps_node-1] [FATAL] [1671498112.357696509] [ublox_gps_node]: Error configuring u-blox: Failed to cold reset device after configuring GNSS
```
**To Reproduce**
Steps to reproduce the behavior:
1. Check out `foxy-devel`
2. Apply the git patch [0001-Edit-ZED-F9P-config-and-add-debug-statements.patch](https://github.com/KumarRobotics/ublox/files/10273220/0001-Edit-ZED-F9P-config-and-add-debug-statements.patch) to the branch `foxy-devel`, which adds additional DEBUG statements, changes the default config to `zed_f9p.yaml`, increases the debug level to 2, and adds the additional config parameters:
```
    gnss:
      glonass: true
      beidou: true
      gps: true
      qzss: true
      galileo: true
      imes: false
```
3. Connect the ZED F9R device via USB, and note the port, which is likely `/dev/ttyACM0`. If not, note the device and edit the config `zed_f9p.yaml` to reflect the proper device.
7. Run the command `ros2 launch unblox_gps ublox_gps_node-launch.py -d`
8. Compare output of the node to the attached log [zed-f9r-foxy-devel-cfg-debug.log](https://github.com/KumarRobotics/ublox/files/10273244/zed-f9r-foxy-devel-cfg-debug.log)
9. See error:
```
[ublox_gps_node-1] [FATAL] [1671498112.357696509] [ublox_gps_node]: Error configuring u-blox: Failed to cold reset device after configuring GNSS
```

**Expected behavior**

- It is expected that a ZED F9R with the ublox_gps node should start publishing ros topics with this config file, which enumerates all the enable/disable parameters for the GNSS configuration.
- It is expected that there is a value for `gnss.qzss_sig_cfg` which will cause the config to be determined to be correct, which there is not. See the log, which shows the intermediate values used in the logic for the `configureUblox()` function: https://github.com/KumarRobotics/ublox/blob/foxy-devel/ublox_gps/src/ublox_firmware8.cpp#L188-L191
- It is expected that reconfiguring the device does not always result in a FATAL

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.5 LTS Focal Fossa (GNU/Linux 5.10.65-tegra aarch64)
 - ROS 2 Foxy
 - NVIDIA Jetson AGX Orin (ARMv8)

vinaykumarhegde, 2022-12-29:
@duxroq Any leads on this? #147 could have been the problem as well.

vinaykumarhegde, 2023-04-03:
Anyone looking at fixing this, two things that worked:
* configure the module on a x86 machine with latest gpsd ubxtool and set `configure_on_boot: false`;
* connecting ublox modules to hardware UART ports on the Orin/ARM module.

Chose the second one and things are working as expected now.