KumarRobotics/ublox #202: Ros2 foxy RTK please

uicuic, 2023-02-09:
I want to run RTK in ros2 foxy with Ublox ZED-F9P. Is there a guide or package that I can use RTK in ros2?

uicuic, 2023-02-09:
I want to run RTK in ros2 foxy with Ublox ZED-F9P. Is there a guide or package that I can use RTK in ros2?
 I try to these pakages.

https://github.com/KumarRobotics/ublox
https://github.com/leighleighleigh/ublox_dgnss
https://github.com/sibalzer/ublox-foxy-RTCM-correction-

 I publish RTCM massages with ros2 ntrip pakage and I checked Ublox gps pakage subscribe RTCM message with rqt_graph. but It failed

fcladera, 2023-02-09:
Are you getting corrections with ntrip? Can you `rostopic echo` those?

You probably have an issue with your topic names

uicuic, 2023-02-10:
I used https://github.com/Road-Balance/ntrip_ros2 to publish RTCM massages and https://github.com/sibalzer/ublox-foxy-RTCM-correction- to subscribe RTCM massages because I can't found subscriber for RTCM massages in this pakage's foxy-devel branch. 
I changed topic names so I was able to see ublox_gps node subscribe RTCM massages.

I have no idea how can I run RTK in this pakage. 
If you know other ntirp pakage that mach with this pakage or guidline for enabling me to write RTK, please let me know.
thanks.

olvdhrm, 2023-02-28:
@uicuic Is it possible to build and run the ros2 branch instead? 

I managed to run this package with a ZED F9P in ROS2, however in the Humble distribution with the ros2 branch. It is able to subscribe to an /rtcm topic where I publish RTCM correction messages from an NTRIP client (https://github.com/LORD-MicroStrain/ntrip_client.git , ros2 branch as well). The only problem with these 2 packages combined is that I needed a /fix to /nmea parses as I was not able to obtain raw /nmea sentence from this package's ros2 branch. 

You can find it here if you'd like to try the configuration: 
https://github.com/olvdhrm/RTK_GPS_NTRIP.git

Hope this helps!