KumarRobotics/ublox #213: Inject speed data for ESF to the F9R

tomlogan501, 2023-04-14:
**Is your feature request related to a problem? Please describe.**
Sometimes the odometer is not wired to the F9R, therefore a software solution should be send to the F9R

**Describe the solution you'd like**
I would like the ROS driver to subscribe to a velocity topic with speed data and send to F9R the required data correctly format as UBX-ESF-MEAS and type 11.
Obviously some parameters should be provided.

**Describe alternatives you've considered**
Compared to a wired version

**Additional context**
ROS2 galactic with feedbacks of ros2 branches

chrissunny94, 2024-03-04:
Hi

tomlogan501, 2024-03-04:
Thanks for your participation.
I don't know long would it be to develop this feature and make it optional also.