KumarRobotics/ublox #220: Ublox F9P could not get RTK solution

yulan0215, 2023-06-01:
**Describe the bug**
A clear and concise description of what the bug is that I used UBlox ZED-F9P and recently I use ntrip caster here: https://github.com/LORD-MicroStrain/ntrip_client/issues/29#issuecomment-1572711569
I can publish the  RTCM msg, but the problem was how can I use my Ublox F9P to use those RTCM messages?
**To Reproduce**
Steps to reproduce the behavior:
1. My RTCM rostopic was: /ntrip_client/rtcm, and I added it here:
I added my RTCM rostopic here: https://github.com/KumarRobotics/ublox/blob/4f107f3b82135160a1aca3ef0689fd119199bbef/ublox_gps/src/node.cpp#L1908 H
However, when I checked my gps rostopic: /ublox_gps/fix, the status was always 0, which means I did not get RTK solution. Do you have any idea about that? Thx!

bnbhat, 2024-08-02:
Hey!

Is this issue resolved? Have you managed to get RTK fix?

I am using ZED-F9P with RTK. It works for me. Try this [NTRIP Client](https://github.com/Lemvos/ntrip_client)

bnbhat, 2025-05-28:
Hey @JungHaeSu 

I have `config_on_startup` set to false and configuration is set from the ucenter 2 software

```yaml
ublox_gps_base_node:
  ros__parameters:
    debug: 0              

    device: /dev/ttyACM0
    frame_id: gps_base

    uart1:
      baudrate: 460800

    config_on_startup: false
    inf: 
      all: true
    publish:
      all: false
```